project-chip/connectedhomeip #34015: [CERT-TEST-FAILURE] Test_TC_IDM_6_2.yaml should expect INVALID_ACTION rather the UNSUSPPORTED_ACCESS and general issue

Apollon77, 2024-06-19:
### Feature Area

Other

### Test Case

Test_TC_IDM_6_2.yaml

### Reproduction steps

The test case has two issues in my eyes:

1.) Step 14 and 16_1 (so the first 16 that comes after 16) should expect returning INVALID_ACTION rather then UNSUPPORTED_ACCESS. According to the specification it should fit in these points:
* A1: Else if reading the event in the path requires a privilege that is not granted to access the cluster in the path, an EventStatusIB SHALL be generated with the UNSUPPORTED_AC­ CESS Status Code. (for concrete path in Test 14)
* A2: If reading the event in the path requires a privilege that is not granted to access the clus­ ter in the path, then the path SHALL be discarded. (for wildcard path in test 16_1)
* B: If no error free existent paths remain, then EventRequests are considered empty. --> leads to empty!
* C: If this action is in response to a Subscribe Request action, ... If both AttributeRequests and EventRequests are empty ... a Status Response Action with the INVALID_ACTION Status Code SHALL be sent to the
initiator, ....

But the test expects UNSUPPORTED_ACCESS which would only happen if "B" is not respected, so it might also be a SDK implementation error in this case.

2.) Steps 15 and later can never be successful because the ACL that are set in Step 16_1 prevent all these from being successful. So it also could be that the step order of the test is wrong in general or some change screwed up the step order.

This is currently blocking a certification!

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

core

Rajashreekalmane, 2024-06-21:
Test-Plan is expecting unsupported access instead of invalid action. 
Refer to the issue for more details: https://github.com/CHIP-Specifications/chip-test-plans/issues/3442.

 Additionally, please find the existing SDK issue here: https://github.com/project-chip/connectedhomeip/issues/29203.

Apollon77, 2024-06-21:
Ok then I would close here because we have an SDK and an Test plan issue already. No need for duplicates

closed 2024-06-21 by Apollon77