project-chip/connectedhomeip #34017: [BUG] Chiptool logs do not contains timestamp

hiltonlima, 2024-06-19:
### Reproduction steps

When running YAML tests using Matter Test Harness, the received chiptool logs do not contain timestamp information

[image]

### Bug prevalence

Always 

### GitHub hash of the SDK that was being used

fb2c0e5c4e4abe3b88893ab123878c8db458e34b

### Platform

raspi

### Anything else?

Related issue: https://github.com/project-chip/certification-tool/issues/287

bzbarsky-apple, 2024-06-20:
@hiltonlima Is this a duplicate of https://github.com/project-chip/connectedhomeip/issues/24701 in that this is a cross-compiled chip-tool?

hiltonlima, 2024-06-20:
> @hiltonlima Is this a duplicate of #24701 in that this is a cross-compiled chip-tool?

Let me check if it's the same case. And I'll let you know here if it's the same problem

hiltonlima, 2024-06-20:
@bzbarsky-apple It's not the same case. We build on the arm64 architecture for arm64. We don't cross-compile it.

bzbarsky-apple, 2024-06-21:
OK.  Well, we should see whether fixing that issue (being worked on) fixes things for you anyway, since it will fix how the logging backend is selected generally....