project-chip/connectedhomeip #34022: [BUG] Can't group control for OpenThread device

myddpp, 2024-06-20:
### Reproduction steps

When testing the wifi device in the same way, group control works.

This is not the case for thread Demokit devices.

Is another procedure required to group control thread devices?

Below is the method I tested.

`0.  pairing
./chip-tool pairing ble-thread 3  hex:0e080000000000010000000300000f35060004001fffe0020811111111222222220708fd1758cddf0caf40051000112233445566778899aabbccddeeff030e4f70656e54687265616444656d6f010212340410445f2b5ca6f2a93a55ce570a70efeecb0c0402a0f7f8 20202021 3842

1. write acl
./chip-tool accesscontrol read acl 3 0
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 4, "authMode": 3, "subjects": [3], "targets": [{"cluster": 6, "endpoint": 1, "deviceType": null}, {"cluster": 8, "endpoint": 1, "deviceType": null}] }]' 3 0

./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 42, "groupKeySecurityPolicy": 0, "epochKey0": "d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1": "d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2": "d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 3 0

./chip-tool groupkeymanagement write group-key-map '[{"groupId": 2, "groupKeySetID": 42, "fabricIndex": 1}]' 3 0

./chip-tool groupkeymanagement read group-key-map    3 0
./chip-tool accesscontrol read acl 3 0

2. create group
Next, program the chip-tool:
./chip-tool groupsettings add-group groupName groupId
./chip-tool groupsettings add-group grp2      0x0002
 
./chip-tool groupsettings add-keysets keyset Id  keyPolicy     validityTime         EpochKey
./chip-tool groupsettings add-keysets 42     0             0x000000000021dfe0   hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf

./chip-tool groupsettings bind-keyset groupId    keysetId
./chip-tool groupsettings bind-keyset 0x0002     42

./chip-tool groupsettings  show-groups
./chip-tool groupsettings  show-keysets

3. 
Next, add bridge endpoint 3 to Group ID 0x0002:
./chip-tool groups add-group GroupID GroupName destination-id endpoint-id-ignored-for-group-commands
./chip-tool groups add-group 0x0002  grp2      3          1

./chip-tool onoff toggle 0xffffffffffff0002 1
`

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

e5e93bf9af4adcb0ed69062a7b3aa7a8111ebeac

### Platform

nrf, core

bzbarsky-apple, 2024-06-20:
@myddpp Are the devices joining the relevant multicast groups?  Do you have access to the device logs?

https://github.com/project-chip/connectedhomeip/issues/27631 talks about some Thread devices not being configured to support _enough_ groups, but certainly says that just one group should work fine.

Mr-Techtron, 2024-09-08:
> @myddpp Are the devices joining the relevant multicast groups?
@myddpp Hi, I am facing the similar issue that the matter-thread light (espressif esp32c6 based matter thread light) is not responding to the group command. I'm able to control the light through the chip-tool but it doesn't respond to group command. How can I know that the devices have joined the relevant multicast group?

Mr-Techtron, 2025-05-25:
@bzbarsky-apple It seems there is some issue with the group control for matter thread devices, as group control works for the matter-wifi devices. Can you tell me if the group control feature works on matter-thread devices?

myddpp, 2025-05-25:
@bzbarsky-apple 
Yes My cases for matter-thread devices.
But I resolve issue, after kernel config change( multi broad cast enable)

KwangseobJeong, 2025-09-04:
I rebuild my kernel with config change( multi broadcast function enable)