project-chip/connectedhomeip #34024: [1.3] [TC-ACL-2,6] Event number is not propagated in yaml test cases

doublemis1, 2024-06-20:
### Reproduction steps

How to reproduce:
1. Using python runner, execute test case ACL 2.5

After moving to run automatic yaml test cases using python runner, reading event with event number param does not work.
Step 5 is failing due to:
`AssertionError: Test step failed Step 5: TH1 reads DUT Endpoint 0 AccessControl cluster AccessControlEntryChanged event`

```
Test step failure in Step 5: TH1 reads DUT Endpoint 0 AccessControl cluster AccessControlEntryChanged event
Flushed pending ack for MessageCounter:127024878 on exchange 14249i
PostProcessCheckStatus.ERROR: The test expectation "value ({'FabricIndex': 1, 'AdminNodeID': None, 'AdminPasscodeID': 0, 'ChangeType': <ChangeTypeEnum.kAdded: 1>, 'LatestValue': {'FabricIndex': 1, 'Privilege': <AccessControlEntryPrivilegeEnum.kAdminister: 5>, 'AuthMode': <AccessControlEntryAuthModeEnum.kCase: 2>, 'Subjects': [4700216444047178644], 'Targets': None}}) == {'AdminNodeID': 4700216444047178644, 'AdminPasscodeID': None, 'ChangeType': 2, 'LatestValue': {'Privilege': 5, 'AuthMode': 2, 'Subjects': [4700216444047178644], 'Targets': None, 'FabricIndex': 1}, 'FabricIndex': 1}" is false
PostProcessCheckStatus.ERROR: The test expectation "value ({'FabricIndex': 1, 'AdminNodeID': 4700216444047178644, 'AdminPasscodeID': None, 'ChangeType': <ChangeTypeEnum.kRemoved: 2>, 'LatestValue': {'FabricIndex': 1, 'Privilege': <AccessControlEntryPrivilegeEnum.kAdminister: 5>, 'AuthMode': <AccessControlEntryAuthModeEnum.kCase: 2>, 'Subjects': [4700216444047178644], 'Targets': None}}) == {'AdminNodeID': 4700216444047178644, 'AdminPasscodeID': None, 'ChangeType': 1, 'LatestValue': {'Privilege': 5, 'AuthMode': 2, 'Subjects': [4700216444047178644], 'Targets': None, 'FabricIndex': 1}, 'FabricIndex': 1}" is false
PostProcessCheckStatus.ERROR: The test expectation "value ({'FabricIndex': 1, 'AdminNodeID': 4700216444047178644, 'AdminPasscodeID': None, 'ChangeType': <ChangeTypeEnum.kAdded: 1>, 'LatestValue': {'FabricIndex': 1, 'Privilege': <AccessControlEntryPrivilegeEnum.kAdminister: 5>, 'AuthMode': <AccessControlEntryAuthModeEnum.kCase: 2>, 'Subjects': [4700216444047178644], 'Targets': None}}) == {'AdminNodeID': 4700216444047178644, 'AdminPasscodeID': None, 'ChangeType': 1, 'LatestValue': {'Privilege': 3, 'AuthMode': 3, 'Subjects': None, 'Targets': None, 'FabricIndex': 1}, 'FabricIndex': 1}" is false
PostProcessCheckStatus.ERROR: The test expects 3 responses but got 4 responses.
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

main

### Platform

core

### Type

Test Improvement

bzbarsky-apple, 2024-06-20:
@doublemis1 Which "python runner" is this?  Is this chip-repl, chip-tool-python, or darwin-framework-tool-python?

This issue needs much more clear "steps to reproduce".

doublemis1, 2024-06-21:
Usage of ReplTestRunner:
https://github.com/project-chip/connectedhomeip/tree/master/scripts/py_matter_yamltests 

I use directly the ReplTestRunner, And I think that Test Harness also test it in that way.

bzbarsky-apple, 2024-06-21:
Yeah, that runner doesn't implement the LastReceivedEventNumber bits in YAML.  That's why we have those tests disabled in that harness:

```
scripts/tests/chiptest/__init__.py:        "Test_TC_ACL_2_5.yaml",            # chip-repl does not support LastReceivedEventNumber : https://github.com/project-chip/connectedhomeip/issues/28884
scripts/tests/chiptest/__init__.py:        "Test_TC_ACL_2_6.yaml",            # chip-repl does not support LastReceivedEventNumber : https://github.com/project-chip/connectedhomeip/issues/28884
```

The actual Test Harness had better not be using chip-repl for yaml testing in general, because there's a whole bunch of stuff it does not support in YAML (see the other bits of the `__init__.py` file there.

bzbarsky-apple, 2024-06-21:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/28884