project-chip/connectedhomeip #34030: [Platform] linux(arm64)

hassan-xy, 2024-06-20:
### Reproduction steps

I run the command ./gn_build.sh
It builds and takes a space of around 30GB while my linux arm64 dev board has only 8 GB. Any ideas as to how I can cross-compile and resolve the issue.

### Platform

other

### Type

Manually tested with SDK

bzbarsky-apple, 2024-06-20:
`./gn_build.sh` builds a whole bunch of different example apps and configurations.  It's supposed to be a "build and test everything" kind of thing, not appropriate for anything like a constrained device.

What are you actually trying to build?

hassan-xy, 2024-06-20:
I am trying to build chip-tool firstly.
Later I would want to test example codes for Linux arm64 arch.
If there is a way to only build for my arch then that would be great.

bzbarsky-apple, 2024-06-20:
If you want to just build chip-tool, I would recommend either 

```
scripts/examples/gn_build_example.sh examples/chip-tool place/to/put/binary
```

or 

```
./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target linux-arm64-chip-tool"
```

All the builds `./gn_build.sh` does are for your arch, but it does them with different compilers/settings...

hassan-xy, 2024-06-21:
[image]
Well, I am not sure how many more times I see this missing dependencies. Maybe I am missing few commands here.

hassan-xy, 2024-06-21:
I tried to install but still I dont have enough space, I only have 8GB eMMC.
Is it possible to build the binary on x86 ubuntu 20 desktop and move to my arch64 board(ubuntu 18)?

hassan-xy, 2024-06-21:
[image]
Well I am running on the x86 desktop for my target board.

bzbarsky-apple, 2024-06-21:
@andy31415 what's the right way to build just chip-tool while cross-compiling?

andy31415, 2024-06-21:
Only clang can cross-compile, hence the error there saying `due to rule ONLY IF `-(clang|nodeps)`.

The target should be `linux-arm64-chip-tool-clang`

hassan-xy, 2024-06-24:
Hi @andy31415 
I tried and I got a diffrent error this time.
[image]
[image]
I have tried different combinations like --out-prefix, --clean but still the same error.

bzbarsky-apple, 2024-06-24:
@hassan-xy You want
```
./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target linux-arm64-chip-tool-clang build"
```

hassan-xy, 2024-06-25:
Well...
[image]
I am cross-compiling for arm64 environment on x86_64 ubuntu 20.04 desktop.

I am also parallely trying to build on the dev board by building inside of a sd card. I am also facing the same python dependencies. Do you guys have a list of all the python packages or versions necessary for build to succeed?

bzbarsky-apple, 2024-06-25:
The sysroot env var presumably needs to be set to the relevant sysroot for the cross-compile that has the other-arhc compilers, libraries, etc?   That's about as much as I know about the cross-compiling bits, which are badly under-documented....

hassan-xy, 2024-07-03:
Hi,
Well I tried to move the binary to my device after building on RPI 4 board but faced issues with glibc 2.35 version and openssl 3.
Even after building the source code to update, the current version I am facing the issue.

So, can you share a list of all the dependencies connectedIP and its example binaries like chip-tool has so I can prepare for it.
Also please provide support for the **linux yocto environment**. I very popular among developers and will beneficial for everyone.

I will be waiting for your responses.
THANKS!!!

hassan-xy, 2024-07-12:
Guys, I am waiting for your comments...

bzbarsky-apple, 2024-07-12:
I think fundamentally: no one knows the answers to your questions.

hassan-xy, 2024-07-12:
Well can you at least share the I'd of the person working with platform integration.

https://www.yoctoproject.org/
Link to the yocto project.
It's just another flavour of the Linux distro but quite popular.

Let me know if anything else is needed.

hassan-xy, 2024-07-30:
Still wait for a comment....

hassan-xy, 2024-08-07:
No updates here as well.

closed 2024-08-07 by hassan-xy