project-chip/connectedhomeip #34041: Create a Second Test Context for TestReadInteraction

Alami-Amine, 2024-06-21:
https://github.com/project-chip/connectedhomeip/blob/9d56786ed093473fdfb66d770a2e2338ab579d3e/src/app/tests/TestReadInteraction.cpp#L232-L233

- before the migration to pw_unit_tests, `TestReadInteraction` Got executed with two contexts with the second test context having `sSyncScheduler = true`. This was not ported as pw_unit_test doesn't offer a clear way to do that and we had to wait for the migration effort to be finished to do that.

- A workaround to do this was mentioned @arkq 

- This workaround should be done after the changes in #34036 are integrated.

Alami-Amine, 2024-08-10:
Hello @arkq,
any chance we can get your above-mentioned workaround?

arkq, 2024-08-12:
Not until pigweed update, which will happen after full migration to ubuntu 24.04.

closed 2024-11-04 by mergify[bot]