project-chip/connectedhomeip #34048: [Test Failed] Flashing and Runtime Errors ESP32-C6-DevKitM-1 Lighting-App Example

juanezete, 2024-06-24:
### Test issue(s)

**Problem**

I have followed all steps to configure and build the Matter lighting-app specifically for ESP32-C6 following the guide [here](https://project-chip.github.io/connectedhomeip-doc/guides/esp32/build_app_and_commission.html). However, I encounter the following errors during the flashing process and runtime:

**1. ESP32 Monitor Logs:**
`I (108680) NimBLE: GATT procedure initiated: indicate;
I (108680) NimBLE: att_handle=18
I (108740) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (111220) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (113720) NimBLE: GATT procedure initiated: indicate;
I (113720) NimBLE: att_handle=18
I (113780) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (116300) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 16
I (118800) NimBLE: GATT procedure initiated: indicate;
I (118800) NimBLE: att_handle=18
I (118910) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 14
I (119360) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 0 ) indicate = 0
I (119360) chip[DL]: CHIPoBLE unsubscribe received
I (119360) chip[BLE]: Releasing end point's BLE connection back to application.
I (121750) chip[DL]: BLE GAP connection terminated (con 0 reason 0x213)
I (121750) app-devicecallbacks: CHIPoBLE disconnected
I (121760) app-devicecallbacks: Current free heap: 192704
`

**2. Linux terminal Logs:**
`[1719223801.303382][30626:30626] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1719223801.303416][30626:30626] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1719223801.303487][30626:30626] CHIP:FP: Shutting down FabricTable
[1719223801.303538][30626:30626] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1719223801.303560][30626:30626] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1719223801.303563][30626:30626] CHIP:TS: Reverted Last Known Good Time to previous value
[1719223801.303659][30626:30627] CHIP:DL: Bluez disconnected
[1719223801.304403][30626:30627] CHIP:DL: Bluez notify CHIPoBluez connection disconnected
[1719223801.303704][30626:30626] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-2giA2p)
[1719223801.305017][30626:30626] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1719223801.305076][30626:30626] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1719223801.305082][30626:30626] CHIP:DL: Inet Layer shutdown
[1719223801.305085][30626:30626] CHIP:DL: BLE Layer shutdown
[1719223801.306361][30626:30626] CHIP:DL: System Layer shutdown
[1719223801.306752][30626:30626] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:619: CHIP Error 0x00000032: Timeout
`
**Additional Information**

-     Configuration Details: I have meticulously configured the build specifically for ESP32-C6 as per the documentation. (I'm using ESP32-C6-DevKitM-1)
-     Operating System: Ubuntu 22.04 running on a virtual machine.
-     Bluetooth Adapter: USB Bluetooth adapter is used for connectivity.
-     Logs: Detailed logs for both ESP32 monitor and Linux terminal are attached in a separate text file for detailed analysis.

[log_monitor.txt](https://github.com/user-attachments/files/15953668/log_monitor.txt)
[log_terminal.txt](https://github.com/user-attachments/files/15953669/log_terminal.txt)

### Platform

esp32

bzbarsky-apple, 2024-06-24:
@dhrishi @shubhamdp

juanezete, 2024-06-27:
I have resolved the issue. The problem was caused by the virtual machine's NAT configuration, which was blocking the commissioning process. Changing the network setting to Bridged Adapter fixed the issue, and everything is now functioning correctly.

closed 2024-06-27 by juanezete