project-chip/connectedhomeip #34067: [Trivial] chip-tool did not have version number or git hash in cli command

resghst, 2024-06-25:
### Type

Other

### Explanation

holp the chip-tool can add chip-tool --version or chip-tool version command.

### Platform

raspi

### Testing

Manually tested with SDK

andy31415, 2024-06-26:
The trick is to decide what version to expose here ... current version command line handling prints out hard-coded text I believe, which is only useful among manjor versions (which we probably could change), however that also creates churn (keep updating the version file).

To avoid churn, we had various requests to use `git` versions/info/logs for this however we intentionally avoided this due to slowing down builds (we try hard to not execute extra programs) and making builds potentially less reproducible (different output on the same code, depending on git checkout status/state).

The closest equivalent is defining an argument and allowing build systems to override it (and builds system may choose to take the extra time to figure out a good tag). `sl_matter_version_str` is an example of this being done ( it is slow when used, adds extra churn to maintain the version in the builder script).

bzbarsky-apple, 2024-06-26:
@resghst Where is the chip-tool coming from whose version you want to know?  Understanding that will make it easier to figure out how best to address this.

resghst, 2024-06-26:
Sometimes, I built the chip-tool a few months ago. I only leave the chip-tool bin file, so I can not easily get the chip-tool version.

For instance, the chip-tool builds from:
- the hash code: b63c271
- the nearest tag: v1.3.0.0
- branch: master

I holp chip-tool version command will show:  master-v1.3.0.0-b63c271
at least show the hash code.

resghst, 2024-06-26:
I think only project-chip git-hash will fine

bzbarsky-apple, 2024-06-26:
@resghst Build it into a directory whose name is the SHA?

resghst, 2024-06-27:
for example, the following command can control the node 1 endpoint 1.
$ ./chip-tool onoff on 1 1
I won get the chip-tool version with
$ ./chip-tool version
it will return "master-v1.3.0.0-b63c271" or "b63c271"

closed 2024-07-11 by resghst