project-chip/connectedhomeip #34076: [Test Failed] Device composition python test times out when sending DataReport with "suppressResponse=true"

Apollon77, 2024-06-25:
### Test issue(s)

Formally the specification mandates to send DataReports with "suppressresponse = true". When this is done then the Python Device Composition test times out. See https://github.com/project-chip/matter.js/actions/runs/9665311907/job/26662207711#step:4:956

### Platform

all

bzbarsky-apple, 2024-06-26:
@Apollon77 Is it possible that you set SuppressResponse=true for an empty report that was part of a Subscribe transaction (priming reports), not a Report transaction (incremental subscription updates)?  That would in fact cause a test failure, I would think, but would also be a spec violation, so the test failure is good.

Apollon77, 2024-06-26:
No, the DeviceComposition python is do ing a */*/* attribute read request and I had "suppressResponse=true" for all datareports in this, see the log link in the first post. Also you can see that chip-sdk was correctly answering then all with standalone acks

Apollon77, 2024-06-26:
Ok, after clarifications in #34077 i close here. All good.

closed 2024-06-26 by Apollon77