project-chip/connectedhomeip #34077: [BUG] SDK does not send Read DataReports with "suppressResponse=true"

Apollon77, 2024-06-25:
### Reproduction steps

Specification tells in (Matter Specs 1.3) in 8.4.3.3. Outgoing Report Data Action

> SuppressResponse MAY be set to TRUE for a Report Data action that initiates a Report transac­tion that conveys an empty list of AttributeReports and EventReports, otherwise:
◦ SuppressResponse SHALL be set to TRUE for a Report Data action that is part of a Read transaction.
◦ SuppressResponse SHALL be set to FALSE for a Report Data action that is part of a Subscribe transaction.

I implemented the first mandate here in matter.js and had a python test failure. See https://github.com/project-chip/connectedhomeip/issues/34076

This error in the python test and the fact that the python tests run with matter,js when using "supporessResponse=false" means that the chip sdk also always sends suppressResponse=false on DataReports (because elso the pythong tests would have failed earlier.

In chip sdk when searching for suppressResponse I can not find any mentioning in the places where DataReports are sent, this means that it is always unset aka false. This in fact violates the specification (or specification needs to be adjusted).

I will revert this change in matter.js for now to stay "comparable to chip".

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

core

bzbarsky-apple, 2024-06-26:
@Apollon77 When you say "first mandate", do you mean this bit:

> MAY be set to TRUE for a Report Data action that initiates a Report transac­tion that conveys an empty list

or this bit:

> SuppressResponse SHALL be set to TRUE for a Report Data action that is part of a Read transaction.

?

Apollon77, 2024-06-26:
I mean the "SHALL" because the "MAY" is just valid for empty data reports how I read it. But in fact when SDK do not "opt in" top send  suppressResponse=true for empty reports (what it does not do when I understood the code right) then the "SHALL" in the next line becomes effective required for the empty Datareport which would mean the "true" again ... so in fact I read this as

* Read DataReport-> required to always send "suppressResponse=true"
* Subscription DataReport -> empty DR can be "suppressResponse=true", else required "=false"

Maybe this part should be rephrased ? ;-)

bzbarsky-apple, 2024-06-26:
What the current spec says (and I agree maybe it could be clearer in terms of phrasing, but it _is_ clearly and unambiuously defined right now) is this:

1. Read DataReport -> SHALL always send "SuppressResponse=true"
2. Subscription DataReport that is not part of the priming sequence and is empty -> MAY send "SuppresResponse=true".

The SDK implements item 1, and never sets SuppressResponse=true for item 2, so the SDK is spec-compliant as far as I can tell here....

bzbarsky-apple, 2024-06-26:
Or put another way, there are three transaction types that can have a ReportData action:

1. Report transaction (incremental subscription update): MAY have the value true if no data being reported.
2. Read transaction: SHALL have the value true.
3. Subscribe transaction (priming reports): SHALL have the value false (because the SubscribeResponse needs to follow the StatusReport).

bzbarsky-apple, 2024-06-26:
And for item 1, the "that initiates a Report transac­tion" wording is _extremely_ important.  It's not all empty data reports.  It's specifically empty data reports that start a new Report transaction.  That is, only things that are doing "8.6.2. Report Transaction Empty" from the main Matter spec.  This is the "we got to max-interval for the subscription and have nothing to report" case.

There can be empty ReportData actions in other cases (wildcard read, subscribe that sent DataVersions that match current state) where this spec clause does not apply at all.

Apollon77, 2024-06-26:
Ok, thats interesting ... I used a real matter device (Onevis) which I assume uses chip sdk. I can also verify with a chip device if you like ....

```
2024-06-26 15:58:10.056 DEBUG  InteractionClient    Sending read request to udp://fddc:3d12:3702:1:5297:c984:d888:279a:5540 on session secure/30654 for attributes */*/* and events undefined
2024-06-26 15:58:10.059 DEBUG  ExchangeManager      Message » id: 44676/36155/108400825 type: 1/2 acked: undefined reqAck: true duplicate: false payload: 153600171818290324ff0b18
2024-06-26 15:58:10.255 DEBUG  MessageExchange      Message « id: 30654/36155/235879994 type: 1/5 acked: 108400825 reqAck: true duplicate: false payload: 1536011535012600de56d41c370124020024031d2504fdff1824020118181535012600de56d41c370124020024031d24040018360215240016240101181818181535012600de56d41c370124020024031d240401183602041d041f0428043004310433043c043e043f0435042a1818181535012600de56d41c370124020024031d24040218360204291818181535012600de56d41c370124020024031d24040318360204011818181535012600de56d41c370124020024031d2504fcff1824020018181535012600de56d41c370124020024031d2504f8ff1836021818181535012600de56d41c370124020024031d2504f9ff1836021818181535012600de56d41c370124020024031d2504fbff18360205f8ff05f9ff05fbff05fdff040004010402040305fcff1818181535012600941b3b26370124020024031f2504fdff1824020118181535012600941b3b26370124020024031f24040018360215240105240202360307248f2869cd0dde7618340424fe01181818181535012600941b3b26370124020024031f2404021824020418181535012600941b3b26370124020024031f2404041824020418181535012600941b3b26370124020024031f2404031824020318181535012600941b3b26370124020024031f2504fcff1824020018181535012600941b3b26370124020024031f2504f8ff1836021818181535012600941b3b26370124020024031f2504f9ff1836021818181535012600941b3b26370124020024031f2504fbff18360205f8ff05f9ff05fbff05fdff040004020404040305fcff18181815350126004c48533237012402002403282504fcff18240200181815350126004c485332370124020024032824040018240201181815350126004c4853323701240200240328240406182c02025858181815350126004c4853323701240200240328240401182c02054f6e766973181815350126004c48533237012402002403282404021825023d14181815350126004c4853323701240200240328240403182c02025334181815350126004c48533237012402002403282404041825020110181815350126004c485332370124020024032824040718240201181815350126004c4853323701240200240328240408182c020131181815350126004c485332370124020024032824040918260290bd9800181815350126004c485332370124020024032824040a182c0209312e312e302d333135181815350126004c485332370124020024032824041318350224000324010318181815350126004c48533237012402002403282504fdff18240201181818290324ff0118
2024-06-26 15:58:10.272 DEBUG  InteractionMessenger Received DataReport chunk with 31 attributes and 0 events, suppressResponse: undefined, moreChunkedMessages: true
2024-06-26 15:58:10.273 DEBUG  ExchangeManager      Message » id: 44676/36155/108400826 type: 1/1 acked: 235879994 reqAck: true duplicate: false payload: 1524000024ff0b18
2024-06-26 15:58:10.486 DEBUG  MessageExchange      Message « id: 30654/36155/235879995 type: 1/5 acked: 108400826 reqAck: true duplicate: false payload: 15360115350126004c4853323701240200240328240405182c02200000000000000000000000000000000000000000000000000000000000000000181815350126004c4853323701240200240328240412182c021031394144373543314644353334304446181815350126004c48533237012402002403282504f8ff18360218181815350126004c48533237012402002403282504f9ff18360218181815350126004c48533237012402002403282504fbff18360205f8ff05f9ff05fbff05fcff040004060401040204030404040704080409040a041305fdff0405041218181815350126009971f45537012402002403302504fcff18240200181815350126009971f455370124020024033024040218240200181815350126009971f455370124020024033024040318240200181815350126009971f455370124020024033024040118350224003c2501840318181815350126009971f4553701240200240330240404182902181815350126009971f45537012402002403302504fdff18240201181815350126009971f455370124020024033024040018240200181815350126009971f45537012402002403302504f8ff18360204010403040518181815350126009971f45537012402002403302504f9ff18360204000402040418181815350126009971f45537012402002403302504fbff18360205f8ff05f9ff05fbff05fcff040204030401040405fdff04001818181535012600f35bb4c837012402002403312404001824020118181535012600f35bb4c837012402002403312404011836021530000820112014020519772901181818181535012600f35bb4c837012402002403312404021824020a18181535012600f35bb4c837012402002403312404031824021418181535012600f35bb4c8370124020024033124040418290218181535012600f35bb4c837012402002403312404051824020018181535012600f35bb4c8370124020024033124040618300208201120140205197718181535012600f35bb4c8370124020024033124040718340218181535012600f35bb4c837012402002403312504fcff1824020218181535012600f35bb4c837012402002403312504fdff1824020118181535012600f35bb4c837012402002403312504f8ff1836020401040504071818181535012600f35bb4c837012402002403312504f9ff183602040004030404040604081818181535012600f35bb4c837012402002403312504fbff1836020400040104020403040404050406040705f8ff05f9ff05fbff05fcff05fdff18181815350126007149efa5370124020024033324040018360218181818290324ff0118
2024-06-26 15:58:10.496 DEBUG  InteractionMessenger Received DataReport chunk with 29 attributes and 0 events, suppressResponse: undefined, moreChunkedMessages: true
2024-06-26 15:58:10.497 DEBUG  ExchangeManager      Message » id: 44676/36155/108400827 type: 1/1 acked: 235879995 reqAck: true duplicate: false payload: 1524000024ff0b18
[... cut ...] 0402401824020018181535012600c56ab73d3701240201240306250403401824020018181535012600c56ab73d37012402012403062504f8ff1836021818181535012600c56ab73d37012402012403062504f9ff1836020400040104020440044104421818181535012600c56ab73d37012402012403062504fbff18360205f8ff05f9ff05fbff05fcff05fdff040005004005014005024005034018181818290424ff0118
2024-06-26 15:58:11.403 DEBUG  InteractionMessenger Received DataReport chunk with 14 attributes and 0 events, suppressResponse: true, moreChunkedMessages: undefined
```

bzbarsky-apple, 2024-06-26:
In the _IM action_ for a read, the logical SuppressResponse boolean is true.

How that gets _encoded_ when chunking is involved is covered by "10.7.3. ReportDataMessage", but now we do have a spec bug: it needs to have some wording about how "this is set to the value in the action, unless MoreChunkedMessages is true, in which case SuppressResponse must be false".  This is obviously the only thing that makes sense, because for chunking you need the response to know when to send the next chunk.

An example that handles this better is "10.7.6. WriteRequestMessage", which says:

> SuppressResponse SHALL NOT be set to true if MoreChunkedMessages is true.

So the SDK behavior here looks correct and we need to fix the encoding spec a bit, right?

cecille, 2024-06-26:
Given that clients do not send responses to either report, does this flag actually convey any actionable information? For interoperability, we want to make sure the reports are processed properly by clients, so maybe we should just clarify the spec to match what clients in the field expect?

closed 2024-06-26 by bzbarsky-apple