project-chip/connectedhomeip #34083: Do we have any example for room air conditioner. If not, then how to create such device

Aryan-Tamboli, 2024-06-26:
### Reproduction steps / Feature

Any resources on how to build room air conditioner device

### Platform

esp32

### Type

Manually tested with SDK

bzbarsky-apple, 2024-06-26:
You use the "Room Air Conditioner" type from the device library, configure that in your ZAP configuration for the device, then enable any optional features you might need.

Past that, you'd need to ask more specific questions, but this does not seem like an issue in the Matter SDK.

closed 2024-06-26 by bzbarsky-apple

Aryan-Tamboli, 2024-06-27:
Thanks for helping. I have generated the ZAP files, now how to integrate it with example and flash the firmware?

bzbarsky-apple, 2024-07-03:
It depends on which example you are using, exactly...