project-chip/connectedhomeip #34084: linux arm64

hassan-xy, 2024-06-26:
### Reproduction steps

Well I am trying to build on the target arm64  board using linux yocto and a second board using ubuntu 18.04 as well. But the python package psutil is a unable to build for me and throws an error. The problem is with the python 3 headers missing.
By the time I resolve this **can anyone please share the chip-tool arm64 binary files with me**. Even if it is incomaptible I would like to give it a try on my system. 

### Platform

other

### Platform Version(s)

linux arm64. (Both yocto and ubuntu 18.04)

### Type

Platform validated

bzbarsky-apple, 2024-06-27:
@andy31415

hassan-xy, 2024-06-27:
Dont bother, I tried to build on rpi 4 ubuntu 22 and migrate it but failed with the openssl3 and glibc dependencies.

hassan-xy, 2024-06-27:
Ohhhh bummer....

I can't update my legacy system to Ubuntu 22.

However, I have a new device that uses Linux Yocto (Dunfell). It's quite popular among devs. Can you list down the system requirements so I can try to hack my way through the build process?