project-chip/connectedhomeip #34085: [Platform] linux

hassan-xy, 2024-06-26:
### Reproduction steps

Hi please share  a list of all the python packages and versions used to build the example codes.

### Platform

python, other

### Type

Manually tested with SDK, Platform validated

bzbarsky-apple, 2024-06-27:
@andy31415

andy31415, 2024-06-27:
An exact list may not be easily built. I would suggest to run a build that you need and then run `pip list` to fetch the list of installed packages.

hassan-xy, 2024-06-27:
I get an error related to psutils package, I tried every article out there but just cant get through the issue, Can you have a look at the log and revert back to me.

Building wheels for collected packages: mobly, psutil, pyperclip
  Building wheel for mobly (setup.py): started
  Building wheel for mobly (setup.py): finished with status 'done'
  Created wheel for mobly: filename=mobly-1.12.1-py3-none-any.whl size=160690 sha256=8f4af6380d5b45b0f351e92e186c332f878d1e0f192370a90e02c081a0404c6d
  Stored in directory: /root/.cache/pip/wheels/63/bb/6f/5332fa07292cfad692d273db4b9d6b140d50913113a03922bf
  Building wheel for psutil (pyproject.toml): started
  error: subprocess-exited-with-error
  
  �× Building wheel for psutil (pyproject.toml) did not run successfully.
  �│ exit code: 1
  �╰�─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for psutil (pyproject.toml): finished with status 'error'
  ERROR: Failed building wheel for psutil
  Building wheel for pyperclip (setup.py): started
  Building wheel for pyperclip (setup.py): finished with status 'done'
  Created wheel for pyperclip: filename=pyperclip-1.9.0-py3-none-any.whl size=11002 sha256=5b8ccaf3cb807f548de558f1b977e1c4e07efaeb26324ff0caec6fe5f6eb687e
  Stored in directory: /root/.cache/pip/wheels/6d/a5/02/e801d8db207899213bc1e564f861696b03cb9598c2817b2b62
Successfully built mobly pyperclip
Failed to build psutil
ERROR: Could not build wheels for psutil, which is required to install pyproject.toml-based projects
ninja: build stopped: subcommand failed.
['ninja', '-C', '/run/media/mmcblk1p1/edge3ConnectedIp/.environment/gn_out', '-v', ':python_packages.install']

Traceback (most recent call last):
  File "/run/media/mmcblk1p1/edge3ConnectedIp/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 421, in install_packages
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/run/media/mmcblk1p1/edge3ConnectedIp/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/run/media/mmcblk1p1/edge3ConnectedIp/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 995, in <module>
    sys.exit(main())
  File "/run/media/mmcblk1p1/edge3ConnectedIp/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 987, in main
    return EnvSetup(**vars(parse())).setup()
  File "/run/media/mmcblk1p1/edge3ConnectedIp/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 577, in setup
    result = step(spin)
  File "/run/media/mmcblk1p1/edge3ConnectedIp/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 766, in virtualenv
    if not virtualenv_setup.install(
  File "/run/media/mmcblk1p1/edge3ConnectedIp/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
  File "/run/media/mmcblk1p1/edge3ConnectedIp/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 424, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['ninja', '-C', '/run/media/mmcblk1p1/edge3ConnectedIp/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.
root@se-704a0efeee54:/run/media/mmcblk1p1/edge3ConnectedIp#

andy31415, 2024-06-27:
I don't seem to see the undelying error from the logs. Try a `pip install psutil==5.9.4` and see if that fails out with a clearer error message.

closed 2024-08-07 by hassan-xy