project-chip/connectedhomeip #34091: [Tooling] API compatibility checker does not consider apiMaturity="provisional" on structs

bzbarsky-apple, 2024-06-26:
### Reproduction steps / Feature

https://github.com/project-chip/connectedhomeip/actions/runs/9686593855/job/26729361125?pr=34090 complains (among other more reasonable complaints):

```
2024-06-26 20:52:39 ERROR   Struct Thermostat::QueuedPresetStruct has been deleted.
```

but the XML said:

```
  <struct name="QueuedPresetStruct" apiMaturity="provisional">
```

### Platform

core (please add to version below)

### Type

Manually tested with SDK

andy31415, 2024-07-08:
The rough bit here is that the compatibility checker compares 'matter' files. While .matter files support provisionality everywhere, the construction of matter files originates from zap, and zap support for provisional bits is limited due to queries needing to be updated one by one (i.e. we do not have provisional tags for structs or other things like events or enums or bitmaps).

So to fix this we need:

  - [ ] add ZAP support for provisional at least in structs
  - [ ] update codegen for matter files to tag structs as provisional
  - [ ] update the compatibility checker to allow changes to provisional structs (we can do delete or more likely allow any changes while the struct is provisional)

andy31415, 2024-07-08:
Change for ZAP in https://github.com/project-chip/zap/pull/1353