project-chip/connectedhomeip #34102: [CERT-TEST-FAILURE] - TC-GRPKEY- 2.2 -Step22 fail

mideayanghui, 2024-06-27:
### Reproduction steps

Use TH Version: TH Fall2023 Sha: 27090a17 to test
Step 22 fail.
If we send the cmd manually.
`./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 3,"groupKeySecurityPolicy": 0, "epochKey0":"d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 1,"epochKey1":"d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 18446744073709551613,"epochKey2":"d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 18446744073709551614 }' 1 0` in Step 22, it actually return RESOURCE_EXHAUSTED as expected.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008

### Platform

other

### Platform Version(s)

ASR

### Type

Common Cluster Logic

### Anything else?

TH Log:
[TC-GRPKEY-2-2_2024_06_27_11_24_04.log](https://github.com/user-attachments/files/16009215/TC-GRPKEY-2-2_2024_06_27_11_24_04.log)

bzbarsky-apple, 2024-06-27:
From the log:

```
INFO       | 2024-06-27 03:27:31.922996 | [2024-06-27 11:26:00] ubuntu@ubuntu:~/apps$ ./chip-tool groupkeymanagement key-set-write {"groupKeySetID": 3, "groupKeySecurityPolicy": 0, "epochKey0": "hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 1, "epochKey1": "hex:d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 18446744073709551613, "epochKey2": "hex:d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 18446744073709551614} 1 0
```

That's missing singly quotes around the write argument, so then everything gets parsed wrong by the shell and things fail, no?

This does not seem like an SDK issue.  Should this actually be filed in https://github.com/CHIP-Specifications/chip-test-scripts/issues ?  Because clearly TH is just doing the wrong thing here.

bzbarsky-apple, 2024-06-27:
@cjandhyala please see above?

cjandhyala, 2024-06-27:
@mideayanghui in step 21 (which is a manual step) did you do keyWrite until the GroupKeyMap attribute contains Max_GrpKey entries , if you don't do that , step 22 will return success but not RESOURCE_EXHAUSTED. pls chk that step from your execution. Since those are manual steps, those are not logged in the log file you attached.