project-chip/connectedhomeip #34107: [Tooling] [chip-cert] CD certs are not properly generated and have encoding errors

andrei-menzopol, 2024-06-27:
### Reproduction steps / Feature

It looks like chip-cert doesn't generate valid CD certs.

The issue was spotted during the run of [TC_DA_1_2.py](https://github.com/project-chip/connectedhomeip/blob/ee49ebdd86669429aa68a8fb5b5c9b756928b9cc/src/python_testing/TC_DA_1_2.py) while parsing the CD certificate [L381](https://github.com/project-chip/connectedhomeip/blob/ee49ebdd86669429aa68a8fb5b5c9b756928b9cc/src/python_testing/TC_DA_1_2.py#L381), however the issue can be reproduced with the example [chip-cert#gen-cd-example](https://github.com/project-chip/connectedhomeip/tree/ee49ebdd86669429aa68a8fb5b5c9b756928b9cc/src/tools/chip-cert#gen-cd-example).

By checking the generated **cd.bin** certificate with **openssl**, it will generate an error. It can also be reproduced with **python module cryptography & x509.load_der_x509_certificate function**.

`unable to load certificate
140226802017600:error:0D0680A8:asn1 encoding routines:asn1_check_tlen:wrong tag:../crypto/asn1/tasn_dec.c:1149:
140226802017600:error:0D07803A:asn1 encoding routines:asn1_item_embed_d2i:nested asn1 error:../crypto/asn1/tasn_dec.c:309:Type=X509_CINF
140226802017600:error:0D08303A:asn1 encoding routines:asn1_template_noexp_d2i:nested asn1 error:../crypto/asn1/tasn_dec.c:646:Field=cert_info, Type=X509`

[image]

The certificates from [credentials/test/certification-declaration](https://github.com/project-chip/connectedhomeip/tree/ee49ebdd86669429aa68a8fb5b5c9b756928b9cc/credentials/test/certification-declaration) also have this issue.

[credentials/development/cd-certs/Chip-Test-CD-Cert.der](https://github.com/project-chip/connectedhomeip/blob/ee49ebdd86669429aa68a8fb5b5c9b756928b9cc/credentials/development/cd-certs/Chip-Test-CD-Cert.der) doesn't have this issue.

### Platform

core (please add to version below)

### Platform Version(s)

ee49ebdd86669429aa68a8fb5b5c9b756928b9cc

### Type

Manually tested with SDK

cecille, 2024-06-27:
The CD is a der file, but it's not an x509 cert. Try asn1parse.

ex:
```
Try cecille@cecille1:~/chip/connectedhomeip$ openssl asn1parse -inform DER -in credentials/development/commissioner_dut/struct_dac_vidpid_fallback_encoding_01/cd.der 
    0:d=0  hl=3 l= 232 cons: SEQUENCE          
    3:d=1  hl=2 l=   9 prim: OBJECT            :pkcs7-signedData
   14:d=1  hl=3 l= 218 cons: cont [ 0 ]        
   17:d=2  hl=3 l= 215 cons: SEQUENCE          
   20:d=3  hl=2 l=   1 prim: INTEGER           :03
   23:d=3  hl=2 l=  13 cons: SET               
   25:d=4  hl=2 l=  11 cons: SEQUENCE          
   27:d=5  hl=2 l=   9 prim: OBJECT            :sha256
   38:d=3  hl=2 l=  68 cons: SEQUENCE          
   40:d=4  hl=2 l=   9 prim: OBJECT            :pkcs7-data
   51:d=4  hl=2 l=  55 cons: cont [ 0 ]        
   53:d=5  hl=2 l=  53 prim: OCTET STRING      [HEX DUMP]:152400012501F1FF360204B118250334122C04135A494732303134315A423333303030312D32342405002406002507769824080018
  108:d=3  hl=2 l= 125 cons: SET               
  110:d=4  hl=2 l= 123 cons: SEQUENCE          
  112:d=5  hl=2 l=   1 prim: INTEGER           :03
  115:d=5  hl=2 l=  20 prim: cont [ 0 ]        
  137:d=5  hl=2 l=  11 cons: SEQUENCE          
  139:d=6  hl=2 l=   9 prim: OBJECT            :sha256
  150:d=5  hl=2 l=  10 cons: SEQUENCE          
  152:d=6  hl=2 l=   8 prim: OBJECT            :ecdsa-with-SHA256
  162:d=5  hl=2 l=  71 prim: OCTET STRING      [HEX DUMP]:304502210080094E7ED4E6513C65AAFC6ED4A62C5AC44B191C0101B3678A3914BE841913C702200642B29F67F492A898EDE088C207A448D108CDC4AE992016D1E0DEC1C599E750
```

andrei-menzopol, 2024-06-28:
Thank you.
And what can I do about `load_der_x509_certificate`?
The TC_DA_1_2.py script breaks when trying to load a CD cert here: [TC_DA_1_2.py#L381](https://github.com/project-chip/connectedhomeip/blob/ee49ebdd86669429aa68a8fb5b5c9b756928b9cc/src/python_testing/TC_DA_1_2.py#L381)

tcarmelveilleux, 2024-07-02:
There are 2 separate things:

1. CD signature verification certificates (i.e. what is loaded at https://github.com/project-chip/connectedhomeip/blob/ee49ebdd86669429aa68a8fb5b5c9b756928b9cc/src/python_testing/TC_DA_1_2.py#L381), which are used to verify CD signatures.
  a. https://github.com/project-chip/connectedhomeip/blob/ee49ebdd86669429aa68a8fb5b5c9b756928b9cc/credentials/development/cd-certs/Chip-Test-CD-Cert.der is such a file, and it's the certificate for the SDK's CD test signing key.
2. The Certification Declaration (CD) files themselves. These are CMS signed enveloped.

I think it is possible that there is a misunderstanding about these different formats, relative tot he issue.

@andrei-menzopol Please can you explain what you are trying to do? It is 100% expected that a `cd.bin` from `chip-cert gen-cd` command will not be loadable with `openssl x509 .....` commands.

andrei-menzopol, 2024-07-03:
Thank you @tcarmelveilleux 

I wanted to run the TC_DA_1_2.py script with a custom nxp generated certificate chain.
I managed to successfully run it now. We are using [this certificate](https://github.com/project-chip/connectedhomeip/blob/ee49ebdd86669429aa68a8fb5b5c9b756928b9cc/scripts/tools/nxp/generate_certs.py#L55) when using development certs. I just had to convert the CD-Signing-Cert in der format and using it in the script

closed 2024-07-03 by andrei-menzopol