project-chip/connectedhomeip #34125: [1.3] Apk Size of Android app is over 500 Mb

swap9391, 2024-06-28:
### Reproduction steps

Apk size of app is getting higher with libs folder, any solution to reduce it. 
I have tried enable proguard 

[image]

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

1.3

### Platform

android

### Platform Version(s)

14

### Type

Core SDK Memory Issue

andy31415, 2024-07-08:
are the .so files stripped in this case? We use templates quite heavily and this could contain a lot of debug information.

Before stripping:
[image]
[image]

After stripping:

[image]
[image]

I am however unsure how you got the 385MB so file though ... that one is very large.

closed 2024-08-13 by yunhanw-google

Trishia, 2024-10-11:
@andy31415 
In my case, `./scripts/build/build_examples.py --clean --target android-arm64-chip-tool-no-debug build` does not make any difference on android library size which is still over 80M.  
How to strip android library (so files) ?  
Advise me plz.