project-chip/connectedhomeip #34128: [CERT-TEST-FAILURE][ TC-ICDM-2.1] - Python script Failed

Rajashreekalmane, 2024-06-28:
### Feature Area

Area:ICDPhase2

### Test Case

TC-ICDM-2.1

### Reproduction steps

In the TC_ICDM_2_1 test case script, an UnboundLocalError occurs due to the variable uatHintInstructionDepedentBitmap being referenced before it is assigned a value. This causes the test to fail

Command used to launch the DUT :` ./lit-icd-app `
Command used to validate the python script : 
` python3 TC_ICDM_2_1.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --storage-path admin_storage.json --paa-trust-store-path ../../credentials/development/paa-root-certs/ --PICS /home/ubuntu/PICS_For_Reference_Platform_RPI_Usedby_SDKQA_V1_3
`

> [MatterTest] 06-28 07:04:16.687 INFO Avahi browse: remove
> [MatterTest] 06-28 07:04:16.690 INFO Found an existing secure session to [1:0000000012344321]!
> [MatterTest] 06-28 07:04:16.693 INFO <<< [E:2757i S:48126 M:266881311] (S) Msg TX to 1:0000000012344321 [D6A7] [UDP:[fe80::a01e:65ff:fe27:256d%veth127bfbd]:5540] --- Type 0001:02 (IM:ReadRequest)
> [MatterTest] 06-28 07:04:16.695 INFO >>> [E:2757i S:48126 M:106430649 (Ack:266881311)] (S) Msg RX from 1:0000000012344321 [D6A7] --- Type 0001:05 (IM:ReportData)
> [MatterTest] 06-28 07:04:16.703 INFO 		***** Test Step 8 : TH reads from the DUT the UserActiveModeTriggerHint attribute.
> [MatterTest] 06-28 07:04:16.703 INFO <<< [E:2757i S:48126 M:266881312 (Ack:106430649)] (S) Msg TX to 1:0000000012344321 [D6A7] [UDP:[fe80::a01e:65ff:fe27:256d%veth127bfbd]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
> [MatterTest] 06-28 07:04:16.703 INFO 		**** Skipping: 8
> [MatterTest] 06-28 07:04:16.706 ERROR Exception occurred in test_TC_ICDM_2_1.
> Traceback (most recent call last):
>   File "/home/ubuntu/Jun24/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 783, in exec_one_test
>     test_method()
>   File "/home/ubuntu/Jun24/connectedhomeip/src/python_testing/matter_testing_support.py", line 1506, in async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/home/ubuntu/Jun24/connectedhomeip/src/python_testing/TC_ICDM_2_1.py", line 226, in test_TC_ICDM_2_1
>     self.set_bits_count(uatHintInstructionDepedentBitmap), 1, "UserActiveModeTriggerHint has more than 1 bit that is dependent on the UserActiveModeTriggerInstruction")
> UnboundLocalError: local variable 'uatHintInstructionDepedentBitmap' referenced before assignment
> [MatterTest] 06-28 07:04:16.714 INFO 		***** Test Failure : 
> [MatterTest] 06-28 07:04:16.714 INFO Finished test in 132ms
> [MatterTest] 06-28 07:04:16.715 INFO [Test] test_TC_ICDM_2_1 ERROR
> 

**PFA log below :** 
[TC-ICDM-2_1.txt](https://github.com/user-attachments/files/16025763/TC-ICDM-2_1.txt)

### Bug prevalence

whenever I do 

### GitHub hash of the SDK that was being used

58536ba64da75f0a63c569baa29966008ffdf306

### Platform

raspi

Rajashreekalmane, 2024-07-01:
The script needs to be updated by adding test step 11, where the TH reads the MaximumCheckInBackoff attribute from the DUT.

Test plan PR:  : [4225](https://github.com/CHIP-Specifications/chip-test-plans/pull/4225)

mkardous-silabs, 2024-08-01:
@Rajashreekalmane Your PICS file does not seem correct. From the error logs, you skip step 8 but not step 9 which not spec compliant.

Rajashreekalmane, 2024-08-09:
script has been updated  recently, hence closing this issue

closed 2024-08-09 by Rajashreekalmane