project-chip/connectedhomeip #34134: [BUG] [Python][Controller] Creating multiple simulated controllers cause not possible to commission over BLE

doublemis1, 2024-06-28:
### Reproduction steps

1. Flash device which need to be commission over BLE,
2. Start BLE advertisement
3. Apply needed changes to provided script
4. Run Provided script -> Commissioning fail

### Bug prevalence

always

### GitHub hash of the SDK that was being used

main

### Platform

python

### Platform Version(s)

1.3

### Anything else?

Script to reproduce the issue and logs from the python chip module can be found in below zip file:
[script_and_logs.zip](https://github.com/user-attachments/files/16028822/script_and_logs.zip)

The python controller fail commissioning in presented setup due to:

```
Traceback (most recent call last):
  File "test.py", line 78, in <module>
    result = controller1.CommissionThread(
  File "/home/michal/.local/lib/python3.8/site-packages/chip/ChipDeviceCtrl.py", line 1826, in CommissionThread
    return self.ConnectBLE(discriminator, setupPinCode, nodeId)
  File "/home/michal/.local/lib/python3.8/site-packages/chip/ChipDeviceCtrl.py", line 440, in ConnectBLE
    self._ChipStack.CallAsync(
AttributeError: 'NoneType' object has no attribute 'raise_on_error'
```

andy31415, 2024-07-08:
@doublemis1 what does `Apply needed changes to provided script` step mean?

doublemis1, 2024-07-08:
in zip file there is a script with the reproduction and need update the discriminator, path to paa discriminator and passcode.
However the issue was visible before:
https://github.com/project-chip/connectedhomeip/pull/33989

Now I am looking at this PR, which handle other issue found on our side:
https://github.com/project-chip/connectedhomeip/pull/34231

I think that we can close this specific issue.

closed 2024-08-08 by doublemis1