project-chip/connectedhomeip #34143: Version validation failed when trying to install CHIP REPL

pavel808, 2024-06-30:
### Reproduction steps

My goal is to install and use CHIP REPL to work with customized cluster servers on my ESP32-C6 device. I am building on Ubuntu 22.04.4 LTS.

I am following the instructions as defined here : https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/python_chip_controller_building.md#building-and-installing

When I reach the following stage : **scripts/build_python.sh -m platform -i separate**

I get the following issue below. I'd appreciate any help with this. 

```
Checking the environment:

20240630 12:35:37 INF Environment passes all checks!

Environment looks good, you are ready to go!

Done. Made 8187 targets from 378 files in 200ms
ninja: Entering directory `./out/python_lib'
[48/1043] ACTION //src/controller/data_model:data_model_zapgen_generate(//build/toolchain/linux:linux_x64_gcc)
FAILED: gen/src/controller/data_model/zapgen/zap-generated/access.h gen/src/controller/data_model/zapgen/zap-generated/gen_config.h gen/src/controller/data_model/zapgen/zap-generated/endpoint_config.h gen/src/controller/data_model/zapgen/zap-generated/IMClusterCommandHandler.cpp 
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../src/controller/data_model --default-toolchain=//build/toolchain/linux:linux_x64_gcc --current-toolchain=//build/toolchain/linux:linux_x64_gcc --capture-output --python-dep-list-files gen/src/controller/data_model/data_model_zapgen_generate_metadata_path_list.txt -- ../../scripts/tools/zap/generate.py --no-prettify-output --templates /home/user/connectedhomeip/src/app/zap-templates/app-templates.json --output-dir /home/user/connectedhomeip/out/python_lib/gen/src/controller/data_model/zapgen/zap-generated --lock-file /home/user/connectedhomeip/out/python_lib/zap_gen.lock --parallel /home/user/connectedhomeip/src/controller/data_model/controller-clusters.zap
Searching for zcl file from /home/user/connectedhomeip/src/controller/data_model/controller-clusters.zap
Checking ZAP from ['/home/user/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/zap/zap-cli']:
  !!! Version validation failed: required at least 2024.6.6, got 2024.3.14 instead
[57/1043] c++ obj/src/lib/support/libSupportLayer.CHIPArgParser.cpp.o
ninja: build stopped: subcommand failed.

```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

28430f1364f541329fc5dd4350495d5afd9b8644

### Platform

esp32

### Type

Spec Compliance Issue, Platform Issue

bzbarsky-apple, 2024-07-03:
@pavel808 Well, what ZAP version are you using?  Are you just using the ZAP from bootstrap, or do you have a separate (older) ZAP installed that you are pointing at via the env vars?

pavel808, 2024-07-03:
@bzbarsky-apple Thanks for your reply. In the end, I installed it from the connectedhomeip inside the esp-matter repository after doing a git pull.

I still can't get this to work though. When I try to send a command I keep receiving this: `ERROR: Avahi resolve failed`

What i'm trying to do. My Node ID is 0x7293. I have a custom cluster server on my device with ID 0x131bfc00. There is a command from that cluster with ID 0x1, and I am trying to send value 0x1 there via the command.  

Would you know the correct format to send such a command from CHIP REPL? Maybe i'm doing something wrong. Thanks.

bzbarsky-apple, 2024-07-03:
I don't really know how to use the Python repl, sorry.  That said, it sounds like there are issues with DNS-SD resolution in your setup....  Hard to say more without knowing what exactly is being resolved and what is being advertised.

pavel808, 2024-07-03:
@bzbarsky-apple Ok thanks. It's a very difficult tool to use. 

Is there any other Matter controller tool than be used to test and send commands to custom cluster server?

chip-tool only works with standard known clusters etc. 

There is the ESP Matter Controller example, but it turns out that this is not supported for the ESP32-C6. I've hit a brick wall it seems:)

bzbarsky-apple, 2024-07-03:
> chip-tool only works with standard known clusters etc.

chip-tool can absolutely do custom things.  See https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md#using-wildcards-with-any-command which talks about wildcards, but you can just use whatever cluster/command/attribute IDs you want.

The value syntax there for commands/writes is JSON.  The built-in docs describe how those payloads work (e.g. if you run `chip-tool any command-by-id` the help output has a long section about the `payload` argument, and `chip-tool any write-by-id` has the same for `attribute-values`).

pavel808, 2024-07-04:
@bzbarsky-apple 

My bad. I hadn't realized this. I tried it and it works great. Thanks a lot.

bzbarsky-apple, 2024-07-05:
@pavel808 You're welcome!

andy31415, 2024-07-08:
Closing: issue was not having a propper zap-cli installed, which should be installed after bootstrap.

closed 2024-07-08 by andy31415