project-chip/connectedhomeip #34149: [Android] Use MEI char array attributes

law-ko, 2024-07-01:
### Reproduction steps

We are trying to gather Manufacturer cluster information based on [samples-app-for-matter-android](https://github.com/google-home/sample-apps-for-matter-android) and attempt to edit the `sample-mei-cluster.xml` to read XML attributes but we are having a hard time to understand what the attribute line should look like, here's our attempt:
```
<attribute side="server" code="0x0000" define="MAC_ADDR" type="chararray" writable="false" default="">MacAddress</attribute>
```

And we are tried to build the lib with `scripts/tools/zap_regen_all.py` but failed. Is there any guide on how to do to read MEI attributes which are char array?

### Platform

android

### Type

Manually tested with SDK

### Anything else?

https://developers.home.google.com/matter/extensions/sample-cluster

bzbarsky-apple, 2024-07-03:
I don't know what you mean by "chararray".  The types in the XML are spec-defined types.  For something like a MAC address, you probably want `octstr` (which represents a byte sequence).

closed 2024-07-03 by bzbarsky-apple

law-ko, 2024-07-03:
@bzbarsky-apple We do not know where to find the lib and import into Android Studio, can you give us a guide on where to do so?