project-chip/connectedhomeip #34170: [Build]

parth-shah-25, 2024-07-03:
### Build issue(s)

Getting below mentioned error while building the **'lighting-app'** project in Visual Studio Code using [ESP-IDF Extension](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension) 

### Error is mentioned below

```
FAILED: gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h gen/app-codegen/cpp-app/app/callback-stub.cpp gen/app-codegen/cpp-app/app/cluster-init-callback.cpp /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/app/callback-stub.cpp /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/app/cluster-init-callback.cpp 
cd /home/meet/connectedhomeip/examples/lighting-app/esp32/build/esp-idf/main && /home/meet/connectedhomeip/scripts/codegen.py --generator cpp-app --output-dir /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app --expected-outputs /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/expected.outputs /home/meet/connectedhomeip/examples/lighting-app/lighting-common/lighting-app.matter
Traceback (most recent call last):
  File "/home/meet/connectedhomeip/scripts/codegen.py", line 28, in <module>
    from matter_idl.matter_idl_parser import CreateParser
ModuleNotFoundError: No module named 'matter_idl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/meet/connectedhomeip/scripts/codegen.py", line 32, in <module>
    from matter_idl.matter_idl_parser import CreateParser
  File "/home/meet/connectedhomeip/scripts/py_matter_idl/matter_idl/matter_idl_parser.py", line 7, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'
[659/1220] Building C object esp-idf/esp_http_client/CMakeFiles/__idf_esp_http_client.dir/esp_http_client.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "ninja" terminated with exit code: 1. 
```

I have followed the documentation present in official ESP-IDF website to set it up. [Followed documentation link](https://github.com/espressif/vscode-esp-idf-extension/blob/master/README.md)

bzbarsky-apple, 2024-07-03:
@parth-shah-25 You need to be using a Python that has lark installed.  That means either the one that's set up during bootstrap (and not using that might be a bug in the esp-idf extension?) or making sure that module is installed in whatever Python you're using.

@andy31415

parth-shah-25, 2024-07-05:
@andy31415  What about this 'matter_idl' not found error. How do I encounter this error?

```
FAILED: gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h gen/app-codegen/cpp-app/app/callback-stub.cpp gen/app-codegen/cpp-app/app/cluster-init-callback.cpp /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/app/PluginApplicationCallbacks.h /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/app/callback-stub.cpp /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/app/cluster-init-callback.cpp 
cd /home/meet/connectedhomeip/examples/lighting-app/esp32/build/esp-idf/main && /home/meet/connectedhomeip/scripts/codegen.py --generator cpp-app --output-dir /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app --expected-outputs /home/meet/connectedhomeip/examples/lighting-app/esp32/build/gen/app-codegen/cpp-app/expected.outputs /home/meet/connectedhomeip/examples/lighting-app/lighting-common/lighting-app.matter
Traceback (most recent call last):
  File "/home/meet/connectedhomeip/scripts/codegen.py", line 28, in <module>
    from matter_idl.matter_idl_parser import CreateParser
ModuleNotFoundError: No module named 'matter_idl'

```