project-chip/connectedhomeip #34174: [1.3] Matter 1.3 Device Commission failed with Alexa

Semi-Bin-Blain, 2024-07-03:
### Reproduction steps

My matter device is a Matter1.3 Plug (Zephyr& Matter over Thread)
It can be commissioned with Apple Home, Google Home, Samsung Smart Things, and chip-tool；
It failed to commission with Alexa:
After scanning the OR code, the App shows that： **This device isn't Matter-compatible** 
Then I chose yes to continue the commissioning work.
After waiting for a while the App showed the commission failed, App showed **Error: RN002**.
I checked the device log, it paused at the BLE communication stage until the app filed then BLE disconnect.

Any suggestions and responses would be greatly appreciated!

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

other

### Platform Version(s)

1.3

### Type

Platform Issue, Core SDK Interopability Issue

### Anything else?

[image]  [image]
[Plug_Device_Alexa_commission_failed.log](https://github.com/user-attachments/files/16082523/Plug_Device_Alexa_commission_failed.log)

bzbarsky-apple, 2024-07-03:
@Semi-Bin-Blain This sounds like an issue you should report to Amazon, not a core SDK problem, at first glance....

chrisdecenzo, 2024-07-03:
Please reach out to someone from Alexa for help on this issue (for example, via CSA slack). I believe this error is returned when a problem is encountered with the DAC, for example, if it is not signed by the PAI.

Semi-Bin-Blain, 2024-07-04:
> @Semi-Bin-Blain This sounds like an issue you should report to Amazon, not a core SDK problem, at first glance....

@bzbarsky-apple  At first, I thought it was an Alexa issue, but I got new info that the commit: 8c9ef988f3c3a3c67b9ba681e82f52ae192c1b31 can work correctly with Alexa, so that makes me confused, I tried to compare with the commit: 8c9ef988f3c3a3c67b9ba681e82f52ae192c1b31, but I do not get any useful info for this problem.
So I am reporting here, also I'm trying to report to Amazon.

Semi-Bin-Blain, 2024-07-04:
> Please reach out to someone from Alexa for help on this issue (for example, via CSA slack). I believe this error is returned when a problem is encountered with the DAC, for example, if it is not signed by the PAI.

Thanks, I will check that way also.

bzbarsky-apple, 2024-07-04:
@Semi-Bin-Blain Yeah, there were a _lot_ of changes between those commits.

That said, if you can confirm that one works and the other does not, then you can `git bisect` to find which exact change made it stop working....

Semi-Bin-Blain, 2024-07-04:
@bzbarsky-apple  Yeah, I tried that, there are too many commits that need to be checked...
 However lucky me,  I noticed one known issue #32875 
 My problem is caused by it also, Following the fix of issue #32875 I figured my issue out. 
 Thanks!

bzbarsky-apple, 2024-07-05:
> Yeah, I tried that, there are too many commits that need to be checked...

That's why bisects are nice.  Even if there are thousands of commits it only takes a low double-digit number of tests....