project-chip/connectedhomeip #34215: [1.3] Missing Definition for Parameter min in the Binding Cluster

agatah2333, 2024-07-05:
### Reproduction steps

The value definition for the parameter min is missing in the Binding cluster.

[image]

the min of group_id should be 1
```
./chip-tool binding write binding '[{ "group":0, "cluster":6}]' 0x654325 0
```
This command could work incorrectly due to the missing constraints.

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

other

### Platform Version(s)

1.3

### Type

Common Cluster Logic, Spec Compliance Issue, Core SDK Performance Improvement, Core SDK Interopability Issue

bzbarsky-apple, 2024-07-06:
This XML format does not have a way to express constraints on struct fields.

That said, the actual implementation in the Binding cluster needs to be enforcing the constraint, and it does not seem to do that.

agatah2333, 2024-07-08:
If you check the other XML files, for example, [application-basic-cluster.xml](https://github.com/project-chip/connectedhomeip/blob/v1.3.0.0/src/app/zap-templates/zcl/data-model/chip/application-basic-cluster.xml#L32C134-L32C158), they all have a clear definition of min.

[image]

And one follow-up question: if the code needs to add the limitation, why does the XML provide the detailed limitation? Does it only affect the .matter file when using ZAP?

bzbarsky-apple, 2024-07-12:
For _writable attributes_ the min value is enforced automatically, at least if the value lives in the Ember datastore.

For _struct fields_ there is no enforcement, because nothing uses those XML values for anything; they are not available from the ZAP database, as far as I know.