project-chip/connectedhomeip #34216: [1.3]

agatah2333, 2024-07-05:
### Reproduction steps

Issue: Incorrect File Name for Scene, Changed to scene manage

The file name for Scene was incorrect and has been changed to scene manage.

https://github.com/project-chip/connectedhomeip/blob/v1.3.0.0/src/app/zap-templates/zcl/data-model/chip/scene.xml

https://github.com/project-chip/connectedhomeip/blob/v1.3.0.0/data_model/clusters/Scenes.xml

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

vscode

### Platform Version(s)

1.3

### Type

Spec Compliance Issue, Core SDK Performance Improvement

bzbarsky-apple, 2024-07-06:
The filename doesn't really matter... What matters is the actual XML.

@lpbeliveau-silabs

agatah2333, 2024-07-08:
I couldn't agree more, but changing the filename may help the spec compliance issue.

bzbarsky-apple, 2024-07-12:
@agatah2333 What exact spec compliance issue?