project-chip/connectedhomeip #34218: [1.3] Inconsistent Description for Flow Measurement, Missing Constraint for Max Value of 0x0002 MaxMeasuredValue

agatah2333, 2024-07-05:
### Reproduction steps

The description for Flow Measurement is inconsistent, and the maximum value for 0x0002 MaxMeasuredValue is missing a constraint.

https://github.com/project-chip/connectedhomeip/blob/v1.3.0.0/src/app/zap-templates/zcl/data-model/chip/flow-measurement-cluster.xml
[image]

[image]

[image]

[image]

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

other

### Platform Version(s)

1.3

### Type

Spec Compliance Issue, Core SDK Interopability Issue

bzbarsky-apple, 2024-07-06:
It's a readonly attribute, so the min/max in the XML are not used for anything, even if they were provided.

closed 2024-07-06 by bzbarsky-apple

bzbarsky-apple, 2024-07-12:
If the ZAP UI uses the value, we could add this, sure.  Does it?