project-chip/connectedhomeip #34224: [Platform] Android phone failed to connect to a FireTV device

Hiwensen, 2024-07-08:
### Reproduction steps

1. Run the tv-casting-app on an Android Pixel 7a device running Android Version 14
2. Run the tv-content-app on a FireTV Stick 4K Max device with Fire OS 7.6.7.3
3. Open the tv-casting-app on the phone, wait for the discovery
4. Click the FireTV device discovered in the list and fail to connect to the device.
Please check the screenshot and logs.

[image]

[connect_failed_logcat.txt](https://github.com/user-attachments/files/16122430/connect_failed_logcat.txt)

### Platform

android

### Platform Version(s)

N/A

### Type

Manually tested with SDK

### Anything else?

Question:
Should I run the tv-platform app on the FireTV device to make the connection work?

Hiwensen, 2024-11-27:
Hi @sharadb-amazon, Do we have any plan to fix this?

Hiwensen, 2024-11-28:
I updated to the latest code with commitId `63182c41e71bbbe` and tried connecting an Android phone to a FireTV device, but the connection still failed. 
Below are the error logs and screenshots:
[image]

[matter_connect_log.txt](https://github.com/user-attachments/files/17945449/matter_connect_log.txt)

pgregorr-amazon, 2024-12-03:
@lazarkov please help answer @Hiwensen questions above regarding the tv-platform app compatibility with different platforms.

lazarkov, 2024-12-03:
> 1. Should I run the tv-platform app on the FireTV device to make the connection work?
> 2. Currently, the tv-casting app can only discover FireTV devices running FireOS 7 and above.
> It can't discover the other FireTV, AndroidTV, and Roku devices, is this the expected behaviour?

1. No. It's sufficient that you install your own content app and that should help you with the connection.
2. Yes. That's the expected behaviour. However once AndroidTV and Roku start running Matter on their production OS then those device will be visible as well.

Hiwensen, 2024-12-09:
@sharadb-amazon Thank you for your reply!
> While reproducing this error, can you upload the logs from the Fire TV side to help us understand why we got the Invalid PASE parameter?

Here are the logs from the FireTV side.
[matter_content_app_logcat.txt](https://github.com/user-attachments/files/18056275/matter_content_app_logcat.txt)

>Note that when using the tv-casting-app, we've usually tested against the sample Linux/Android tv-app.

I will test on a ATV device and get back to you later.

Hiwensen, 2024-12-09:
@pgregorr-amazon 
> The next step should be accepting the connection request on the tv-app/Fire TV. This does not happen automatically on the first connection attempt.

I remember that once during the connection, a PIN code input dialog appeared on FireTV. After entering the PIN code, I didn’t see any response. Later, when connecting again, the dialog no longer appeared.

Below are the display pages of the casting app on the Android phone and the content app on FireTV.
[image]
[image]

Hiwensen, 2024-12-10:
@sharadb-amazon 
> Note that when using the tv-casting-app, we've usually tested against the sample Linux/Android tv-app.

Unfortunately, after running the TV-platform and TV-content apps on an AndroidTV mailbox running Android 9 and an AndroidTV mailbox running Android 11, the phone's casting app cannot discover either of them.
I can share the logs from the TV and phone here. For now, I think we can focus on the connection on FireTV.

pgregorr-amazon, 2024-12-31:
@Hiwensen can you please confirm that we can close this issue?
Resolution:
Casting from the example tv-casting-app to a FireTV requires the developer's Amazon CustomerID to be allow listed for DAC Attestation step in the cloud service.

Hiwensen, 2025-01-10:
> @Hiwensen can you please confirm that we can close this issue? Resolution: Casting from the example tv-casting-app to a FireTV requires the developer's CustomerID to be allow listed for DAC Attestation step in the cloud service.

Yes, we can close this issue, thank you for your help with the connection and adding the CustomerID to be the allow listed.

xqb1, 2025-07-28:
Have you solved this problem? I am also facing this problem now.

xqb1, 2025-07-29:
> [@Hiwensen](https://github.com/Hiwensen)您能确认我们可以解决这个问题吗？ 解决方案： 从示例 tv-casting-app 投射到 FireTV 需要开发者的客户 ID 被列入云服务中 DAC 认证步骤的允许名单。

Hello, I would like to ask how to add the customer ID to the DAC permission list?