project-chip/connectedhomeip #34278: [1.3] Inconsistent Description for Occupancy Sensing cluster

agatah2333, 2024-07-10:
### Reproduction steps

The description for Occupancy Sensing cluster is inconsistent with the spec.

[image]

https://github.com/project-chip/connectedhomeip/blob/v1.3.0.0/src/app/zap-templates/zcl/data-model/chip/occupancy-sensing-cluster.xml#L43
[image]

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

5bb5c9e23d532cea40476fc0bd1d3008522792ba

### Platform

core

### Platform Version(s)

1.3

### Type

Spec Compliance Issue

tcarmelveilleux, 2024-08-27:
Fixing in 1.4

tcarmelveilleux, 2024-08-30:
Fixed by https://github.com/project-chip/connectedhomeip/pull/35220

closed 2024-08-30 by tcarmelveilleux