project-chip/connectedhomeip #34297: [CERT-TEST-FAILURE] - TC-DEM-1.1 - Test failed

KishokG, 2024-07-11:
### Feature Area

Other

### Test Case

TC-DEM-1.1

### Reproduction steps

1. Bring up the DUT with energy-management-app.
2. Pair the DUT using chip-tool through "Onnetwork" transport.
3. Run the command `./chip-tool deviceenergymanagement read cluster-revision 1 1`

**Expected Outcome:**

> [1720690440.539] [50435:50437] [TOO] Endpoint: 1 Cluster: 0x0000_0098 Attribute 0x0000_FFFD DataVersion: 521028924
> [1720690440.539] [50435:50437] [TOO]   ClusterRevision: 4
> [1720690440.540] [50435:50437] [EM] <<< [E:5099i S:17862 M:203971832 (Ack:69830139)] (S) Msg TX to 1:0000000000000001 [C5DD] [UDP:[fe80::e65f:1ff:fe0f:2878%eth0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)

**Actual Outcome:**

> [1720690440.539] [50435:50437] [TOO] Endpoint: 1 Cluster: 0x0000_0098 Attribute 0x0000_FFFD DataVersion: 521028924
> [1720690440.539] [50435:50437] [TOO]   ClusterRevision: 3
> [1720690440.540] [50435:50437] [EM] <<< [E:5099i S:17862 M:203971832 (Ack:69830139)] (S) Msg TX to 1:0000000000000001 [C5DD] [UDP:[fe80::e65f:1ff:fe0f:2878%eth0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)

### Bug prevalence

Whenever I try to read ClusterRevision.

### GitHub hash of the SDK that was being used

6e31453ed626768f9c74bc9e59ffae7a7137999f

### Platform

raspi

### Anything else?

In SPEC, the ClusterRevision value has been updated to 4. As per spec changes, SDK(energy-management-app) has to update the ClusterRevision value from 3 to 4.

**Specification References:** https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/DeviceEnergyManagement.adoc#revision-history

cjandhyala, 2024-08-19:
@KishokG is this still an issue ? Can we close this ticket. ?

KishokG, 2024-08-20:
Yes @cjandhyala It's yet to be fixed

jamesharrow, 2024-08-26:
Note that we have removed TC DEM 1.1 https://github.com/CHIP-Specifications/chip-test-plans/pull/4555 
The cluster revision however has been fixed in the SDK in another SDK PR:
https://github.com/project-chip/connectedhomeip/pull/35067

PR #35067 (2024-08-29): Address various DEM testing issues
Addresses various DEM testing issues:

- Fixes #34954
- Fixes #34960 
- Fixes #34962
- Fixes #34297

- https://github.com/CHIP-Specifications/chip-test-plans/issues/4415
- https://github.com/CHIP-Specifications/chip-test-plans/issues/4482

- Adds a new TC_DEM_2.10.py script for checking Q Quality and that MatterReporting callback is called when expected (found and fixed some issues here!)

This should be cross-reviewed against the DEM Test Plan change PR: 
https://github.com/CHIP-Specifications/chip-test-plans/pull/4543
closed 2024-08-29 by mergify[bot]