project-chip/connectedhomeip #34326: Matter Door lock with HomeKey Support

bilalmalik76, 2024-07-14:
### Feature description

We have developed a Matter door lock and now we want to add support of Homekey by using an iPhone to tap to unlock the door with a home key, something similar to this one.
https://smartapfel.com/matter-door-locks-can-be-unlocked-via-apple-home-key/

Can you please guide how can we achieve this? or if this support is by default available in matter enabled door lock?

### Platform

esp32

adminy, 2024-07-18:
I think the apple home key is not part of matter standard and its used with homekit api not matter api. But it would be cool to have this be part of the matter standard.

adipierro, 2024-09-22:
Seems like Aliro protocol will have something useful for this, see https://github.com/kormax/aliro

bilalmalik76, 2024-10-06:
@adipierro I wonder how they aded Apple home key support in Aqara U300 Matter door lcok
https://www.youtube.com/watch?v=quwKM2BgQFw