project-chip/connectedhomeip #34367: [TC-RVCCLEANM-1.2] In the RVC Clean Mode Cluster, the SupportedModes label has to be updated as per the specification.

Ashwinigrl, 2024-07-17:
### Reproduction steps

1. Execute the below mentioned command to put DUT into a commissionable state :
`./chip-all-clusters-app`
2. Once DUT reach the commissionable state sent below mentioned command on TH.
`./chip-tool pairing onnetwork 1 20202021`
3. Once after successful pairing provide the command:
`./chip-tool  rvccleanmode read supported-modes 1 1`

**Actual Outcome:**

> [1721211500.456] [14118:14120] [TOO] Endpoint: 1 Cluster: 0x0000_0055 Attribute 0x0000_0000 DataVersion: 1384436692
> [1721211500.457] [14118:14120] [TOO]   SupportedModes: 3 entries
> [1721211500.457] [14118:14120] [TOO]     [1]: {
> [1721211500.457] [14118:14120] [TOO]       Label: Vacuum
> [1721211500.457] [14118:14120] [TOO]       Mode: 0
> [1721211500.457] [14118:14120] [TOO]       ModeTags: 1 entries
> [1721211500.457] [14118:14120] [TOO]         [1]: {
> [1721211500.457] [14118:14120] [TOO]           Value: 16385
> [1721211500.457] [14118:14120] [TOO]          }
> [1721211500.457] [14118:14120] [TOO]      }
> [1721211500.457] [14118:14120] [TOO]     [2]: {
> **[1721211500.457] [14118:14120] [TOO]       Label: Wash**
> **[1721211500.457] [14118:14120] [TOO]       Mode: 1**
> **[1721211500.457] [14118:14120] [TOO]       ModeTags: 1 entries**
> **[1721211500.457] [14118:14120] [TOO]         [1]: {**
> **[1721211500.457] [14118:14120] [TOO]           Value: 16386**
> **[1721211500.457] [14118:14120] [TOO]          }**
> [1721211500.457] [14118:14120] [TOO]      }
> [1721211500.457] [14118:14120] [TOO]     [3]: {
> [1721211500.457] [14118:14120] [TOO]       Label: Deep clean
> [1721211500.457] [14118:14120] [TOO]       Mode: 2
> [1721211500.457] [14118:14120] [TOO]       ModeTags: 2 entries
> [1721211500.457] [14118:14120] [TOO]         [1]: {
> [1721211500.457] [14118:14120] [TOO]           Value: 7
> [1721211500.458] [14118:14120] [TOO]          }
> [1721211500.458] [14118:14120] [TOO]         [2]: {
> [1721211500.458] [14118:14120] [TOO]           Value: 16384
> [1721211500.458] [14118:14120] [TOO]          }
> [1721211500.458] [14118:14120] [TOO]      }

**Expected outcome :**

> [1721211500.456] [14118:14120] [TOO] Endpoint: 1 Cluster: 0x0000_0055 Attribute 0x0000_0000 DataVersion: 1384436692
> [1721211500.457] [14118:14120] [TOO]   SupportedModes: 3 entries
> [1721211500.457] [14118:14120] [TOO]     [1]: {
> [1721211500.457] [14118:14120] [TOO]       Label: Vacuum
> [1721211500.457] [14118:14120] [TOO]       Mode: 0
> [1721211500.457] [14118:14120] [TOO]       ModeTags: 1 entries
> [1721211500.457] [14118:14120] [TOO]         [1]: {
> [1721211500.457] [14118:14120] [TOO]           Value: 16385
> [1721211500.457] [14118:14120] [TOO]          }
> [1721211500.457] [14118:14120] [TOO]      }
> [1721211500.457] [14118:14120] [TOO]     [2]: {
> **[1721211500.457] [14118:14120] [TOO]       Label: Mop**
> **[1721211500.457] [14118:14120] [TOO]       Mode: 1**
> **[1721211500.457] [14118:14120] [TOO]       ModeTags: 1 entries**
> **[1721211500.457] [14118:14120] [TOO]         [1]: {**
> **[1721211500.457] [14118:14120] [TOO]           Value: 16386**
> [1721211500.457] [14118:14120] [TOO]          }
> [1721211500.457] [14118:14120] [TOO]      }
> [1721211500.457] [14118:14120] [TOO]     [3]: {
> [1721211500.457] [14118:14120] [TOO]       Label: Deep clean
> [1721211500.457] [14118:14120] [TOO]       Mode: 2
> [1721211500.457] [14118:14120] [TOO]       ModeTags: 2 entries
> [1721211500.457] [14118:14120] [TOO]         [1]: {
> [1721211500.457] [14118:14120] [TOO]           Value: 7
> [1721211500.458] [14118:14120] [TOO]          }
> [1721211500.458] [14118:14120] [TOO]         [2]: {
> [1721211500.458] [14118:14120] [TOO]           Value: 16384
> [1721211500.458] [14118:14120] [TOO]          }
> [1721211500.458] [14118:14120] [TOO]      }

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

05e4c10358a18f9c9295c2dcd7cdd880502d539a

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

As per the specification, the **RVC Clean Mode Cluster** has derived cluster tag values defined as **`DeepClean(0x4000), Vacuum(0x4001), Mop(0x4002)`**. When I send the `Read SupportedModes attribute` command to the DUT, the response should include the supported modes labeled as `DeepClean(0x4000), Vacuum(0x4001), and Mop(0x4002)` instead of `DeepClean(0x4000), Vacuum(0x4001), and Wash(0x4002)`. Therefore, the Mode labels in the Sample app need to be updated according to the specification.

**Step that require review:**
Test step 2

**Specification Reference:**
- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Mode_RVCClean.adoc#72-mode-tags

[image]

**Test plan References:**
- https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/mode_rvc_clean.adoc

bzbarsky-apple, 2024-07-17:
There is some _serious_ confusion here.

A ModeOptionStruct has a mode id (Mode) and a Label (whatever the mode should be called for user purposes).

It also has a _list_ of semantic tags, which have values and names.

So you can have a mode labeled "Special Wash" which has the 0x4000 and 0x4002 tags (indicating DeepClean and Mop semantics).  Or that more could be labeled "Deep Wash".  Or "Deep Mop".  Or "Rumpelstiltskin", if you think that's what will tell the users what the mode does.

In any case, the Label for the ModeOptionStruct has no reason to be equal to the name of any of the mode _tags_, and can't be equal to them in general because there is a _list_ of tags, not just one.

bzbarsky-apple, 2024-07-17:
Also, the "Test plan References:" text above is useless, since it just links to the test plan.  If you think some part of the test plan supports this issue, please reference that _exact_ part of the test plan @Ashwinigrl, so it can be fixed.

closed 2024-07-17 by bzbarsky-apple