project-chip/connectedhomeip #34383: [BUG] Failure to timeout after 20 iterations of incorrect passcode

j-ororke, 2024-07-18:
### Reproduction steps

**Pre-Requisites:**
- TH_CR1 = Test harness as Commissioner 1
- TH_CR2 = Test harness as Commissioner 2
- DUT = Commissionee

**Steps to Reproduce:**
1. Commission DUT to TH_CR1
2. TH_CR1 reads the BasicCommissioningInfo attribute from the General Commissioning cluster on EP0 and saves the MaxCumulativeFailsafeSeconds as timeout
3. TH_CR1 sends an OpenCommissioningWindow command to the DUT with the CommissioningTimeout set to timeout
4. TH_CR2 attempts to establish a PASE connection to the DUT 20 times, using a valid onboarding payload with an incorrect setupcode.
5. TH_CR2 attempts to establish a PASE to the DUT using the correct onboarding payload.

**Expected Results:**
Commissioning attempt should fail with timeout status: 
- Failed to establish secure session to device: src/controller/python/ChipDeviceController-ScriptPairingDeviceDiscoveryDelegate.h:59: CHIP Error 0x00000032: Timeout

**Actual Result:**
When running test without adding any sleep time between iteration 20 and iteration 21 we get the following unexpected error code and output:
- [MatterTest] 07-17 22:54:20.802 ERROR PASESession timed out while waiting for a response from the peer. Expected message type was 33
- Failed to commission: src/controller/python/ChipDeviceController-ScriptDevicePairingDelegate.cpp:89: CHIP Error 0x00000003: Incorrect state

**Log snippet of issue:**
```
07-18 00:02:08.137 ERROR PASESession timed out while waiting for a response from the peer. Expected message type was 33
07-18 00:02:08.138 DEBUG SecureSession[0xffffa406eb30]: Released - Type:1 LSID:39910
07-18 00:02:08.138 ERROR Secure Pairing Failed
07-18 00:02:08.139 INFO Commissioning complete done. Successful? False, errorcode = src/controller/python/ChipDeviceController-ScriptDevicePairingDelegate.cpp:89: CHIP Error 0x00000003: Incorrect state
07-18 00:02:08.141 ERROR Exception occurred in test_TC_ADMIN_1_9.
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/mobly/base_test.py", line 783, in exec_one_test
    test_method()
  File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 1547, in async_runner
    return asyncio.run(runner_with_timeout)
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
    return fut.result()
  File "/root/python_testing/scripts/sdk/TC_CADMIN_1_9.py", line 105, in test_TC_ADMIN_1_9
    await self.CommissionAttempt(params, expectedErrCode=0x32)
  File "/root/python_testing/scripts/sdk/TC_CADMIN_1_9.py", line 79, in CommissionAttempt
    asserts.assert_true(errcode.sdk_code == expectedErrCode, 'Unexpected error code returned from CommissioningComplete')
  File "/usr/local/lib/python3.10/dist-packages/mobly/asserts.py", line 353, in assert_true
    fail(msg, extras)
  File "/usr/local/lib/python3.10/dist-packages/mobly/asserts.py", line 475, in fail
    raise signals.TestFailure(msg, extras)
mobly.signals.TestFailure: Details=Unexpected error code returned from CommissioningComplete, Extras=None

```
**Notes:**
Attempted this testing in a automated manner with following python3 test script: 
- https://github.com/project-chip/connectedhomeip/pull/34342 

### Bug prevalence

Frequent

### GitHub hash of the SDK that was being used

11f94c383171b709712106362e902f65253f4968

### Platform

raspi

### Anything else?

Attached debug log from recent test attempt:
[test_log.DEBUG.txt](https://github.com/user-attachments/files/16276071/test_log.DEBUG.txt)

bzbarsky-apple, 2024-07-18:
The C++ SDK does in fact time out, as you can see from the log.  So this issue seems to be on the client/test side, no?

And nothing in the spec defines how the client should handle a PASE attempt timeout.

cecille, 2024-07-18:
re-opening this - this was meant to track the controller side. The error is getting swapped from timeout to internal somewhere.

bzbarsky-apple, 2024-07-19:
Well, that happens in `src/controller/python/ChipDeviceController-ScriptDevicePairingDelegate.cpp`, sort of:
```
void ScriptDevicePairingDelegate::OnStatusUpdate(DevicePairingDelegate::Status status)
{
...
    case DevicePairingDelegate::Status::SecurePairingFailed:
...
            mOnPairingCompleteCallback(ToPyChipError(CHIP_ERROR_INCORRECT_STATE));
```

Of course that code is not handed the status that would tell it how or why things failed.

bzbarsky-apple, 2024-07-19:
The `OnPairingComplete` call does happen with `CHIP_ERROR_TIMEOUT`, I would think.... but that happens after the OnStatusUpdate, I would also think, and therefore the error that's handed is not propagated because `expectingPairingComplete` got set to false.