project-chip/connectedhomeip #34386: Unable build the Android app in the Mac OS[1.3]

uiabhishek15, 2024-07-18:
### Reproduction steps

Can anyone give the steps to build the Android app in the MAC OS

### Bug prevalence

daily

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip.git

### Platform

android

### Type

Platform Issue

joonhaengHeo, 2024-07-29:
@uiabhishek15 
I successfully built the build using the Build Script in the guide below. Is there anything that doesn't work?
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md#building-android-chiptool-from-scripts

yunhanw-google, 2024-10-08:
it works in my side as well. closing this one, feel free to reopen if the issue is still outstanding.

closed 2024-10-08 by yunhanw-google