project-chip/connectedhomeip #34409: [Quest] Android how to check matter device support Thread?

freemanspace, 2024-07-19:
### Test issue(s)

when i create NetworkCredentials can use
network:NetworkCredentials? = NetworkCredentials.forWiFi(NetworkCredentials.WiFiCredentials(wifi.ssid, wifi.password))
or
network:NetworkCredentials? = NetworkCredentials.forThread(NetworkCredentials.ThreadCredentials(thread.operationalDataset))

**but how to check matter device is support thread Network?**

please help thx

### Platform

android

joonhaengHeo, 2024-07-29:
@freemanspace 
After receiving this callback method, You can check if device is supported Thread or not. (Wi-Fi).
https://github.com/project-chip/connectedhomeip/blob/13fe13ae3b61a5b9adb3b1ab8248fa3db11dddf2/src/controller/java/src/chip/devicecontroller/ChipDeviceController.java#L1815

Afterwards, you can update your Credentials using the method below.
https://github.com/project-chip/connectedhomeip/blob/13fe13ae3b61a5b9adb3b1ab8248fa3db11dddf2/src/controller/java/src/chip/devicecontroller/ChipDeviceController.java#L566

freemanspace, 2024-07-29:
thx a lot ~~~~~~~

closed 2024-10-24 by joonhaengHeo