project-chip/connectedhomeip #34432: Support for Ubuntu 24.04 LTS

baylf2000, 2024-07-21:
### Feature description

Right now building on Ubuntu 24.04 fails in multiple places, as has been mentioned elsewhere. Ubuntu 22.04 is several years old now, and there have been many important improvements between the two LTS versions. Many organizations have already started migrating their workstations to 24.04, and holding that back because of this project is not popular.

I would hope that the changes required to add support for 24.04 are not too complex. It would be a great help to many if support for 24.04 could be added asap. Even if official support is not made available, some information on workarounds to allow 24.04 to be used would be greatly appreciated.

### Platform

all

### Platform Version(s)

NA

bzbarsky-apple, 2024-07-22:
> Right now building on Ubuntu 24.04 fails in multiple places, as has been mentioned elsewhere.

Links to the elsewhere?  Is there a clear list of the things that don't work on Ubuntu 24.04?  If not, providing such a list in this issue would help with getting it resolved.

baylf2000, 2024-07-25:
@bzbarsky-apple Thanks for the reply!

I believe most are Python errors that begin with Ubuntu 23.04, which adds the `EXTERNALLY-MANAGED` marker to it's python environment.  See:

[https://packaging.python.org/en/latest/specifications/externally-managed-environments/#externally-managed-environments](url)

I think it's probably more sensible for someone involved in this project to spin up a 24.04 environment and investigate the issues, as the log output is significant and may not make a lot of sense out of context.

bzbarsky-apple, 2024-07-26:
@andy31415 this sounds like something you would know something about....

micwill755, 2024-07-31:
Hi, having the same issue with 24.04, is there any date when 1.3 will build for 24.04?

baylf2000, 2024-08-04:
I can report that I was able to successfully utilise this project on a fresh install of Fedora 40 (which contains even newer versions of most packages than Ubuntu 24.04) without any real issues, so it does seem to be related to the "EXTERNALLY-MANAGED" Python environment in the latest versions of Ubuntu, as I mentioned above.

micwill755, 2024-08-06:
That's great I will try Fedora 40, and yes you are correct, when I try to bootstrap the Matter SDK by running:
`
source ./scripts/bootstrap.sh`

the following error occurs:

subprocess.CalledProcessError: Command '['gn', 'gen', '/home/mcwlm/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/mcwlm/connectedhomeip/third_party/pigweed/repo"']' died with <Signals.SIGABRT: 6>.
Installing pip requirements for all...
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

rosahay-silabs, 2024-08-07:
python-3.12 is not supported as I know, has to do with the changes that it has brought in the packages. If you can use some other version of python, then it should typically work on any distro.

baylf2000, 2024-08-07:
> python-3.12 is not supported as I know, has to do with the changes that it has brought in the packages. If you can use some other version of python, then it should typically work on any distro.
@rosahay-silabs 

I don't think it's an issue with the version of Python. Fedora 40 uses 3.12, and my tests as mentioned above show it has zero issues, at least as far as bootstrapping the install and completing a simple build.

As I said in my original post, the issue appears to be related to Ubuntu (from 23.04 onward) implementing an `EXTERNALLY-MANAGED` Python environment, where the distro attempts to enforce all python package installs via apt by blocking pip commands without a special flag, unless inside a virtual environment. This means that pip commands outside of a virtual environment will always fail, and apparently this SDK uses those extensively.

There is a detailed description of the changes in my message above. (https://github.com/project-chip/connectedhomeip/issues/34432#issuecomment-2249299691)

jamesharrow, 2025-03-12:
Given that the CI runs on Ubuntu 24.04 (as far as I understand) and this works fine for me (recent install in Feb '25) I suggest we close this issue?

jamesharrow, 2025-03-12:
Closing issue as stale - please raise a new issue with logs if you are still facing issues.

closed 2025-03-12 by jamesharrow