project-chip/connectedhomeip #34452: [Feature] Describe "endpoint added" scenario in spec

j0tunn, 2024-07-23:
### Feature description

For now the spec contains only notes about "power on" scenario for different clusters. 
But in some cases the behavior on scenario "endpoint added" should be different from the scenario "power on".

For example: OTA happens and then 3 days later a new endpoint that has an On/Off cluster is added. What is the expected behavior of StartUpOnOff Attribute in such case?

Related issue: https://github.com/project-chip/connectedhomeip/pull/34160

### Platform

all

bzbarsky-apple, 2024-07-23:
@j0tunn Spec issues should be filed in https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/

j0tunn, 2024-07-24:
> @j0tunn Spec issues should be filed in https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/

Yep. I saw such links, but got 404 trying to go there. Thought it was migrated elsewhere, but I don't know where.

bzbarsky-apple, 2024-07-26:
It just means you don't have permissions to that repository... If you're a CSA member, please reach out to @clapre about access.  If not, please let me know and I can file the issue for you.

j0tunn, 2024-07-26:
@bzbarsky-apple No, I'm not a CSA member.

bzbarsky-apple, 2025-04-15:
@j0tunn Sorry for the lag here....  This somehow got lost.  I filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/11558 to track this.