project-chip/connectedhomeip #34551: [1.3] OnWithTimedOff 0x42 Command Parameter Issues

agatah2333, 2024-07-27:
### Reproduction steps

The OnWithTimedOff 0x42 command's Parameter 1 is of type Bitmap (limited to 0-1), but sending a value of 2 is still executable. Additionally, Parameters 2 and 3 have a constraint with a maximum value of 0xFFFE, but a value of 65535 (0xFFFF) still works:

```
./chip-tool any command-by-id 0x6 0x42 '{"0":2,"1":65535,"2":65535}' 0x654324 1
```
or
```
./chip-tool any command-by-id 0x6 0x42 '{"0":3"1":65535,"2":65535}' 0x654324 1
```

[image]

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

561d23d0db215a99705ff0696e73853c8edf11b2 

### Platform

core

### Platform Version(s)

1.3

### Type

Spec Compliance Issue

bzbarsky-apple, 2024-07-27:
@jmartinez-silabs

jmartinez-silabs, 2024-08-23:
some additional info from slack conversations

> Rob Houtepen
>   In ZCL, this is better described.
> There bitmaps are defined using 'reserved' for unused bits.
> Then there is spec text that tell that on transmission reserved bits shall be 0, and on reception reserved bits shall be ignored.
> So in the case of OnWithTimedOff command in Matter, the constraint of '0 to 1' for the OnOffControl field is misleading. When received a value of 2, it should not lead to a constraint error. But a compliant Matter device is not allowed to send 2 

...

jmartinez-silabs, 2025-04-09:
Not an issue. See previous message. Chip-tool, as a testing device, is allowed to send reserved values.

closed 2025-04-09 by jmartinez-silabs