project-chip/connectedhomeip #34554: [1.3] The Commands in the Level Control Cluster Do Not Obey the Constraint Ranges

agatah2333, 2024-07-27:
### Reproduction steps

OptionsMask and OptionsOverride in OptionsBitmap with Bitmap Parameter Type (Limited to 0-1)

[image]

However, in the commands: MoveToLevel, Move, Step, Stop, and their variants with OnOff, the commands work normally when setting OptionsMask and OptionsOverride to 3.

```
./chip-tool any command-by-id 0x8 0x1 '{"0":1,"1":1,"2":3,"3":3}' 0x654324 1
```

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

561d23d0db215a99705ff0696e73853c8edf11b2 

### Platform

other

### Platform Version(s)

1.3

### Type

Spec Compliance Issue

bzbarsky-apple, 2024-07-27:
@jmartinez-silabs

jmartinez-silabs, 2024-10-15:
I have an open PR to fix this. However not that value **3** is valid for both fields ( bit 1 | bit 0)

> the commands work normally when setting OptionsMask and OptionsOverride to 3.

ReneJosefsen, 2024-10-22:
I agree, the constraints for bitmap is a bit different than other fields, since if we throw a constraint error if would not work well with future clients which does know and sets any added bits. So the server rightfully should ignore the unknown bits. 

On the topic of adding such a constraint to chip-tool, I think if we are using in the context of testing it might be useful to be able to set values outside the allowed range to test for these cases as well. So I'm tempted to state that this is as designed.

bzbarsky-apple, 2024-11-01:
chip-tool must be able to send invalid values, for testing purposes.  When compiling it, we even override some normal checks the SDK does on outgoing data to allow said invalid values to be sent.

jmartinez-silabs, 2025-06-04:
Feedback provided.  Chip-tool, as a testing tool, can send parameters outside the constraint range. Level control server has been updated in 1.4 to check all parameters constraints

closed 2025-06-04 by jmartinez-silabs