project-chip/connectedhomeip #34556: Add support for multiple presets of a given type in the thermostat app for linux

nivi-apple, 2024-07-27:
Currently we are limiting our implementation to supporting 1 preset of each type. The spec allows for having multiple presets.

Also Make sure the ApplyPendingPresets API provides a unique preset handle for the new presets that are added.
Currently we use the octet string containing the presetscenario for the new preset handle. This is unique in our current implementation since currently we only support one preset of each type.

ReneJosefsen, 2024-08-19:
This is not an urgent thing for 1.4 SVE, so this can be pushed to 1.5 as an enhancement. @ReneJosefsen will follow up to make sure this is labelled properly.

nivi-apple, 2024-08-23:
We also need to modify the check for number of presets to include the below - If, after applying all pending changes to the the <<ref_Presets>> attribute, the total number of
<<ref_PresetStruct, presets>> in <<ref_Presets>>, having a <<ref_PresetStructPresetScenario>> matching
<<ref_PresetTypeStructPresetScenario>> in the <<ref_PresetTypes>> attribute, would exceed the value specified in the
<<ref_PresetTypeStructNumberOfPresets>> for that scenario in the <<ref_PresetTypes>> attribute, a response with the
status code RESOURCE_EXHAUSTED SHALL be returned.

ReneJosefsen, 2024-08-26:
> We also need to modify the check for number of presets to include the below - If, after applying all pending changes to the the <<ref_Presets>> attribute, the total number of <<ref_PresetStruct, [...]

That sounds like something that should be added to the test plan and test scripts, right?

ReneJosefsen, 2024-08-26:
HVAC TT (08/26-2024): The above comment should be moved to a TP issue (@ReneJosefsen will raise this and link to the SDK issue)

nivi-apple, 2024-08-27:
Also fix test TC_TSTAT_4_2 - step 18 to account for number of presets of each type. Review comment to be addressed - https://github.com/project-chip/connectedhomeip/pull/35141#discussion_r1733444138

ReneJosefsen, 2024-09-09:
This was closed by mistake, reopening.

nivi-apple, 2025-05-22:
@ReneJosefsen should we close this since our app is an example app. do we need to support multiple presets?

ReneJosefsen, 2025-05-23:
I guess it depends if we plan on doing a test case for this 🤔 I have no strong opinion on this. We already have a TP issue tracking this, so I'm fine with closing this and reopen or create a new one when relevant?

PR #35291 (2024-09-02): Fix remaining post-TE2 TC-SWTCH issues
- Fixes #34556 (-2.2, -2.3 and -2.4 to better match test plan)
- Fixes #28620 (Better bound checks)
- Fixes #35241