project-chip/connectedhomeip #34558: [Platform] Linux Yocto, kernel 5.15

hassan-xy, 2024-07-27:
### Reproduction steps

I am able to run chip tool somehow on my device. I have checked all the necessary configrautions and tried to solve it to the best of my ability.
I am facing issue with adding matter device through chip-tool cli. I tried to use the 
1. ./chip-tool pairing ble-wifi 0x1234 ssid abcd 10101011 1000 but some reason I would see pairing timeout.
2. ./chip-tool pairing onnetwork 0x11 10101011. I have attached screenshots for this.
[image]

[image]

Also notice why time shows **2023**. My RTC is prefectly synced with the correct UTC time.
And why do I get a **mdns timeout error** I checked and my mdns service runs perfectly well.

I am using Linux yocto based device. Please guide me. I have been at this stage since 1 month....
Thanks

### Platform

other

### Type

Platform validated

bzbarsky-apple, 2024-07-29:
@hassan-xy Well, is the device being paired on-network or not?  It sure doesn't look like it from the onnetwork log, and the fact that you were trying ble-wifi also suggests that it's not...  Is it?

hassan-xy, 2024-07-30:
Well it's not pairing currently. I am not sure if bluez stack is to blame. But if it was the onnetwork should work. 
I have added an mDNS service and noted it's working successfully. The Bluetooth service also works fine. Now I am unable to debug this issue. Do you have any logging feature to debug this process?

Any other suggestions you might have.

bzbarsky-apple, 2024-07-30:
I suggest looking at logs from both sides.  But the first step is really figuring out which scenario you are in: are you on-network or not?

hassan-xy, 2024-07-30:
I tried for both and both actually failed.

hassan-xy, 2024-07-30:
Do you have any links to refer to for validating the ipv6 or ble protocol?

bzbarsky-apple, 2024-07-30:
No, I mean which one is the device?

If the device is already on the IP network but on-network pairing is failing, then I would suggest looking at the log to see which part is failing (DNS-SD discovery or something later).  And perhaps comparing to what avahi is reporting for the DNS-SD bits.

If the device is not on the IP network, I have no idea what you should do to debug bluez, sorry.

andy31415, 2024-10-21:
@hassan-xy could you please provide a reproduction script for this one?

Something line an existing sample application + a shell script that uses chip-tool to reproduce the issue. Without that, it seems to me that this is some environment-specific issue that you encounter and then I cannot validate if this is a real SDK issue or not.

hassan-xy, 2024-10-22:
Hi @andy31415 Thanks for looking into the issue. Unfortunately I am working on few other priority issues and will come back to this later.
But since you mention that this could be an issue of the environment and not the SDK let me just ensure we are on the same page. 
I am unable to pull the SDK and build on my target device(linux yocto kirkstone 4.0) due to python software issue(I raised tickets as well for the same but got no update), So I got the chip-tool and binaries from the RPI board for arm64 arch. and used it in my board.
The only other issue I faced was with the libstdc++ linking. So I pulled the needed file from RPI and used a script to change the linking 

#!/bin/bash
cd /usr/lib/
mkdir backup_libs
cp libstdc++.so* backup_libs/.
cp ~/libstdc++.so.6* .
ln -sf libstdc++.so libstdc++.so.6.0.30
cd
./chip-tool
And then it starts running.

Let me know your thoughts.
Thanks

andy31415, 2024-10-24:
We cannot support all linux environments currently  - I am not familiar with yocto linux. I don't think we would expect things build for RPi to run out of the box on other systems. I would suggest to try to setup a sysroot that is yocto-specific to compile. https://github.com/project-chip/connectedhomeip/blob/master/integrations/docker/images/stage-1/chip-build-crosscompile/start-sysroot-vm.sh has the example of building the sysroot for RPi.

if you already have a yocto device, maybe you could construct a sysroot out of it (copying the libs and includes from it into a separate folder) and then cross compile.