project-chip/connectedhomeip #34567: How manual pairing code is generated

jarvissd, 2024-07-29:
### Reproduction steps

Hi everyone, i have a task about understanding matter protocol.
During pair process, i don't understand how manual pairing code is generated base on vendorID, productID,...
i read the document but still don't get it
Can someone give me an explanation about this? thank a lot!!!

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

all-cluster-app and lighting-app for esp32

### Platform

esp32

### Type

Test Improvement

QuentinCaldeira-eaton, 2024-07-29:
Hi @jarvissd , I think you might check the 
`Matter-1.3-Core-Specification, Chapter 5. Commissioning`
Available [here](https://csa--iot-org.translate.goog/developer-resource/specifications-download-request/?_x_tr_sl=en&_x_tr_tl=fr&_x_tr_hl=fr&_x_tr_pto=sc).
I think this should answer your question, in
`5.1.4 Manual Pairing Code` on page 236.

jarvissd, 2024-07-29:
i tried to follow this instruction from document to gen manual code 
but when i tried to decode manual code: 3497 011 2332 ( manual code default) i didn't get right information 
like: vendor ID, discrimiator, product ID,...

QuentinCaldeira-eaton, 2024-07-29:
Which value you want to find vor Vendor ID, Discriminsator, Passcode, and VID_PID_PRESENT ? Also, is your decoding a script or a manual application. Maybe check the `src/setup_payload/Setup_Payload.cpp` will help you

jarvissd, 2024-07-29:
Version:             0
Vendor ID:           65521    (0xFFF1)
ProductID:           32768    (0x8000)
Custom flow:         0        (STANDARD)
Discovery Bitmask:   0x02     (BLE)
Long discriminator:  3840     (0xf00)
Passcode:            20202021
these default informations i tried on examples lighting-app 
and manual code  is 3497 011 2332 
also with QR code MT:Y.K9042C00KA0648G00 
i tried to decode it thought base 38 but it still wasn't right 
@QuentinCaldeira-eaton 
can u explain for me pls :<

QuentinCaldeira-eaton, 2024-07-29:
```python

0001 0011 0100 0100 0010 0010 0101 //PASSCODE 20202021

		    1111 0000 0000 // DISCRIMINATOR 3840

	       1111 1111 1111 0001 // VENDOR ID 0xFFF1

### First digit

VID_IS_PRESENT Seems to be 0, so it will be DISCRIMINATOR >> 10,
so 1111 0000 0000 >> 10 = 0b11, or 3 in decimal

### Second to sixth digit

((DISCRIMINATOR & 0x300) << 6) | (PASSCODE & 0x3FFF)

DISCRIMINATOR & 0x300 = 0b0011 0000 0000 << 6 = 0b0000 1100 0000 0000 0000

PASSCODE & 0x3FFF = 0b0000 0000 0000 0000 0010 0010 0101

so...

  0000 0000 0000 0000 0010 0010 0101 | 
  0000 0000 0000 1100 0010 0010 0101 = 

  0000 0000 0000 1100 0010 0010 0101 or 49701 in decimal

### Seventh to thenth digit

PASSCODE >> 14 = 0b0000 0100 1101 0001 so 1233 in decimal

## Last digit

Follow an algorithm, don't have the time to help you on this one

```

So, the encoding is working, and so on for the decoding. If you want to decode, follow the same path, but invert. Please be aware of logical expression, and use a proper converter to pass from binary to decimal. Is this helpful enough ?

jarvissd, 2024-07-29:
ok thanks for all

closed 2024-07-29 by jarvissd