project-chip/connectedhomeip #34573: [Feature][Darwin] Question of EventNameForID function plan

joonhaengHeo, 2024-07-29:
### Feature description

Are there any plans to support a function that can return EventName with Event ID similar to the function below?
https://github.com/project-chip/connectedhomeip/blob/13fe13ae3b61a5b9adb3b1ab8248fa3db11dddf2/src/darwin/Framework/CHIP/MTRClusterNames.h#L43C88-L43C109

### Platform

darwin

bzbarsky-apple, 2024-07-29:
We could add something like this.  The only reason it's not there so far is that there weren't any known use cases...

joonhaengHeo, 2024-07-29:
@bzbarsky-apple 
I needed this, so I implemented it separately and used it. Is it okay if I contribute it?

bzbarsky-apple, 2024-07-30:
@joonhaengHeo Absolutely, yes!

joonhaengHeo, 2024-07-31:
@bzbarsky-apple 
Thanks. I upload new PR about this.

PR #34653 (2024-07-31): [Darwin] Implement MTREventNameForID function
Fix #34573 

Implement MTREventNameForID function in darwin platform.
Additionally, the return value of MTRAttributeNameForID was incorrect, so this Issue was also fixed.
closed 2024-07-31 by mergify[bot]