project-chip/connectedhomeip #34574: src/credentials/CHIPCertFromX509.cpp:442: Error 0x00000305

SmartHua, 2024-07-29:
### Reproduction steps

class MyKeypairDelegate : KeypairDelegate {
    override fun generatePrivateKey() {

    }

    override fun createCertificateSigningRequest(): ByteArray {
        return CHIPConstant.createCertificateSigningRequest
    }

    override fun getPublicKey(): ByteArray {
        return CHIPConstant.keyPairPublicKey
    }

    override fun ecdsaSignMessage(message: ByteArray): ByteArray {
        return CsrManager.signer(message)
    }
}

 
val myKeypairDelegate = MyKeypairDelegate()
 val operationalKeyConfig = OperationalKeyConfig(
                    myKeypairDelegate,
                    CHIPConstant.rca,
                    CHIPConstant.ica,
                    CHIPConstant.noc,
                    CHIPConstant.ipk
            )
chipDeviceController =
                    ChipDeviceController(ControllerParams.newBuilder(operationalKeyConfig)
                            .setControllerVendorId(VENDOR_ID).build())

How to implement KeypairGate？

### Bug prevalence

/

### GitHub hash of the SDK that was being used

1.2

### Platform

android

sdrsdr, 2024-07-29:
same here with commit 5e3127f5ac61e13c572a968199280d90a9c19dce  chip-cert works for paa but not for pai

sdrsdr, 2024-07-30:
I'm having this on Linux (Ububtu 24.04) so it's not android specific

lmapii, 2024-11-01:
I'm seeing the same when using `chip-cert print-cert` on a `PAI`:

```
chip-cert print-cert credentials/test/attestation/Chip-Test-PAI-FFF2-NoPID-Cert.der
ConvertX509CertToChipCert() failed
src/credentials/CHIPCert.cpp:929: Error 0x00000305
```