project-chip/connectedhomeip #34583: [CERT-TEST-FAILURE]- [TC-SWTCH-2.4] -- Python Script Failure

Ashwinigrl, 2024-07-29:
### Feature Area

Other

### Test Case

TC-SWTCH-2.4

### Reproduction steps

1. Command used to launch the DUT : `./chip-all-clusters-app `
2. Command used to run the python script : `python3 TC_SWTCH.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --endpoint 1 --storage-path admin_storage.json  --paa-trust-store-path ../../credentials/development/paa-root-certs/ --tests test_TC_SWTCH_2_4`

While executing the **TC-SWTCH-2.4** python script, Test case is failing due to the `skip_all_remaining_steps was provided with invalid starting_step_num`.  Line number 242 `skip_all_remaining_steps` function has string argument it needs to be updated to integer.

> [MatterTest] 07-29 10:38:04.833 ERROR Exception occurred in test_TC_SWTCH_2_4.
> Traceback (most recent call last):
>   File "/home/ubuntu/master_CNTRL/connectedhomeip/no/lib/python3.10/site-packages/mobly/base_test.py", line 783, in exec_one_test
>     test_method()
>   File "/home/ubuntu/master_CNTRL/connectedhomeip/src/python_testing/matter_testing_support.py", line 1647, in async_runner
>     return _async_runner(body, self, *args, **kwargs)
>   File "/home/ubuntu/master_CNTRL/connectedhomeip/src/python_testing/matter_testing_support.py", line 1635, in _async_runner
>     return asyncio.run(runner_with_timeout)
>   File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
>     return loop.run_until_complete(main)
>   File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
>     return future.result()
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
>     return fut.result()
>   File "/home/ubuntu/master_CNTRL/connectedhomeip/src/python_testing/TC_SWTCH.py", line 242, in test_TC_SWTCH_2_4
>     self.skip_all_remaining_steps("2")
>   File "/home/ubuntu/master_CNTRL/connectedhomeip/src/python_testing/matter_testing_support.py", line 1127, in skip_all_remaining_steps
>     asserts.fail("skip_all_remaining_steps was provided with invalid starting_step_num")
>   File "/home/ubuntu/master_CNTRL/connectedhomeip/no/lib/python3.10/site-packages/mobly/asserts.py", line 475, in fail
>     raise signals.TestFailure(msg, extras)
> mobly.signals.TestFailure: Details=skip_all_remaining_steps was provided with invalid starting_step_num, Extras=None
> [MatterTest] 07-29 10:38:04.844 INFO 		***** Test Failure : 

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

5e3127f5ac61e13c572a968199280d90a9c19dce

### Platform

raspi

### Anything else?

**Python Script References:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_SWTCH.py#L108

[image]

**Reference Log:**
[TC-SWTCH-2.4.txt](https://github.com/user-attachments/files/16411970/TC-SWTCH-2.4.txt)

tcarmelveilleux, 2024-07-30:
The test has to be run against Endpoint 3 and 4 as well. Please don't only test on Endpoint 1.

cecille, 2024-07-30:
Please update to ToT, this issue shoudl resolve itself. This is one of the new experimental tests, so it doesn't need an endpoint arg. If you don't actually have a button attached (you probably don't). Please either say you're the CI by supplying the ci pics file for QA testing, or append the flag `--bool-arg use_button_simulator: True`

tcarmelveilleux, 2024-08-14:
Fixed in master before TE2