project-chip/connectedhomeip #34594: In fact, there is no such file of `src/platform/OpenThread/BUILD.gn`. Maybe we can add the source set later but not in this PR.

dhrishi, 2024-07-29:
In fact, there is no such file of `src/platform/OpenThread/BUILD.gn`. Maybe we can add the source set later but not in this PR.

_Originally posted by @wqx6 in https://github.com/project-chip/connectedhomeip/pull/33872#discussion_r1675381200_

dhrishi, 2024-07-29:
@wqx6 PTAL

wqx6, 2024-07-30:
I cannot add a source_set for OpenThreadBorderRouter delegate in `src/platform/OpenThread/BUILD.gn`. The sources use the headers such as `platform/ThreadStackManager.h` which are in the static_library `${chip_root}/src/platform:platform`. If we make the thread_br source_set depend on the static_library `${chip_root}/src/platform:platform`. There will be a dependency cycle error. I guess that is the reason why we didn't create BUILD.gn file in `src/platform/OpenThread` before.
```
ERROR Dependency cycle:
  //third_party/connectedhomeip/src/platform:platform ->
  //third_party/connectedhomeip/src/platform/ESP32:ESP32 ->
  //third_party/connectedhomeip/src/platform/OpenThread:border-router-delegate ->
  //third_party/connectedhomeip/src/platform:platform
```