project-chip/connectedhomeip #34620: thread device pairing Tip on ios: The home network must include a Thread border router

Wuou, 2024-07-30:
### Reproduction steps

When using the MatterSupport and Matter frameworks on iOS to pair with a Thread device, always receive the message: "The home network must include a Thread border router." What configuration do I need to do?

I am able to pair with Matter over Wi-Fi devices successfully, why can't I do the same with Thread devices?

### Bug prevalence

Always come up

### GitHub hash of the SDK that was being used

Matter.framework

### Platform

darwin

### Type

Platform Issue

aajain-com, 2024-07-30:
@Wuou  are you trying to add the accessory to Apple Home or to another ecosystem? Please file an Apple Feedback Assistant ticket and attach the complete diagnostic logs and share the feedback ticket number here so I can follow up.

Wuou, 2024-07-31:
> @Wuou are you trying to add the accessory to Apple Home or to another ecosystem? Please file an Apple Feedback Assistant ticket and attach the complete diagnostic logs and share the feedback ticket number here so I can follow up.

Thank you for your reply.

1. We have developed our own app, using MatterSupport and Matter.framework for provisioning Matter devices. This is not Apple Home.
2. We have a Thread border router (using a Raspberry Pi) and a Thread Thermostat device.
3. Provisioning of Wi-Fi devices works normally, but when pairing a Thread device, we consistently receive the message "The home network must include a Thread border router."
4. We have already ensured that the mobile device is using the Raspberry Pi's network.
So, how should we configure the Thread device so that pairing can be successful?
[image]

Wuou, 2024-07-31:
> @Wuou are you trying to add the accessory to Apple Home or to another ecosystem? Please file an Apple Feedback Assistant ticket and attach the complete diagnostic logs and share the feedback ticket number here so I can follow up.

I obtained network information from the thread network selection method in MatterExtension:

override func selectThreadNetwork(from threadScanResults: [MatterAddDeviceExtensionRequestHandler.ThreadScanResult]) async throws -> MatterAddDeviceExtensionRequestHandler.ThreadNetworkAssociation

I found only this network: [networkName: MyHome, extendedPANID: 1744289600-2862571479831496853]. So, is it because the iPhone cannot recognize the Raspberry Pi's network?

JoeyYang2010, 2024-08-22:
@Wuou I am developing my own app too, I have problems when pairing a wifi sensor, "MatterAddDeviceExtensionRequestHandler" will not invoked, can you help me with that, thanks a lot.

woody-apple, 2025-04-02:
Please file a feedback ticket request here:
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple

closed 2025-04-02 by woody-apple