project-chip/connectedhomeip #34646: [BUG] Q color control

cecille, 2024-07-30:
### Reproduction steps

Getting more reports than expected on transitions. looks like 1 per second even after transition is complete. Or something.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

3263e403469fd66827d923484a50ddb504544112

### Platform

core

bzbarsky-apple, 2024-07-30:
@jmartinez-silabs

jmartinez-silabs, 2024-07-30:
@cecille Depending on the transition time, the report interval could be 1s. This should be the smaller interval of report posible. Other than end of transition report. 

Can you clarify/what step we're done to see report after transition? That should not be possible. the new code is only exercise on the calculation of the steps.

I'll have to look at it on Monday, I am still away

PR #34820 (2024-08-07): Fixes for the color control  quiet reporting
fixes #34618
fixes #34646
fixes #34643

- MarkAttributeDirty::kYes instead of kIfChanged for QuietReporting to ensure the value is reported. 
- Fix Start Or End of transition Quiet report to only report if the attribute value changed since the last quiet report
- apply same tweak for start/end of the transition to the lvl control
- re-enable the disabled tc_cc_2_2 steps

Tested with fully enabled TC_CC_2.2.py
closed 2024-08-07 by jmartinez-silabs