project-chip/connectedhomeip #34660: [BUG] Unable to see mDNS information on network, or commissioning

QuentinCaldeira-eaton, 2024-07-31:
### Reproduction steps

I have an app made with the Zephyr platform, and try to perform an Ethernet (On Network) commissioning. As it was discussed before, we don't have to use Zephyr DNS-SD implementation because minimal MDNS already does the job. So, in the platform, CONFIG_NET_L2_ETHERNET (zephyr variable to use ethernet) select chip_mdns_minimal.

My board is a nucleo h563zi, but it only uses Zephyr SDK, no other things, so the app should be easily port. I validate that my board is well configured and capable of pusblish mDNS service with a Zephyr example. I fix the service name to matterc, and use `Resolve-Dnsname matterc ` on PowerShell to discover it. Here is the result : 

[image]

It seems that my board and my network are well configured. After that, I run my Matter application. In the logs, it says 

```
I: Init CHIP stack
I: 5060 [SVR]Subscription persistence not supported
I: 5064 [SVR]Server initializing...
D: 5068 [FP]Initializing FabricTable from persistent storage
I: 5073 [TS]Last Known Good Time: 2023-10-14T01:16:48
I: 5078 [DMG]AccessControl: initializing
I: 5081 [DMG]Examples::AccessControlDelegate::Init
I: 5086 [DMG]AccessControl: setting
I: 5089 [DMG]DefaultAclStorage: initializing
I: 5093 [DMG]DefaultAclStorage: 0 entries loaded
D: 5097 [IN]UDP::Init bind&listen port=5540
E: 5101 [IN]IPV6_PKTINFO failed: 109
D: 5105 [IN]UDP::Init bound to port=5540
D: 5108 [IN]TransportMgr initialized
I: 5112 [ZCL]Using ZAP configuration...
I: 5116 [DMG]AccessControlCluster: initializing
I: 5120 [ZCL]Initiating Admin Commissioning cluster.
D: 5125 [DL]Boot reason: 1
D: 5127 [ZCL]GeneralDiagnostics: OnDeviceReboot
D: 5131 [DMG]Endpoint 0, Cluster 0x0000_0033 update version to 7752e26
D: 5138 [EVL]LogEvent event number: 0x00000000001C0000 priority: 2, endpoint id:  0x0 cluster id: 0x0000_0033 event id: 0x3 Sys timestamp: 0x0000000000000045
E: 5151 [ZCL]DeviceInfoProvider is not registered
D: 5156 [DMG]Endpoint 1, Cluster 0x0000_0004 update version to 98fa3fcc
D: 5162 [DMG]Endpoint 1, Cluster 0x0000_0004 update version to 98fa3fcd
I: 5168 [ZCL]0x8077fcb ep 1 clus 0x0000_0062 attr 0x0000_0000 not supported
D: 5175 [ZCL]ERR: setting LastConfiguredBy on Endpoint 1 Status: 86
D: 5181 [IN]SecureSession[0x20001990]: Allocated Type:1 LSID:58726
D: 5187 [SC]Assigned local session key ID 58726
D: 5191 [SC]Waiting for PBKDF param request
I: 5195 [DIS]Updating services using commissioning mode 1
I: 5201 [DIS]CHIP minimal mDNS started advertising.
E: 5205 [DIS]Failed to get primary mac address of device. Generating a random one.
I: 5212 [DIS]Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1 cp=0
D: 5222 [DIS]Responding with _matterc._udp.local
D: 5226 [DIS]Responding with 64F386FC49D49A8B._matterc._udp.local
D: 5232 [DIS]Responding with 4C3AD5EF3008.local
D: 5236 [DIS]Responding with 4C3AD5EF3008.local
D: 5240 [DIS]Responding with _V65521._sub._matterc._udp.local
D: 5246 [DIS]Responding with _S15._sub._matterc._udp.local
D: 5251 [DIS]Responding with _L3840._sub._matterc._udp.local
D: 5256 [DIS]Responding with _CM._sub._matterc._udp.local
D: 5262 [DIS]Responding with 64F386FC49D49A8B._matterc._udp.local
I: 5267 [DIS]CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 64F386FC49D49A8B.
I: 5277 [DIS]mDNS service published: _matterc._udp
I: 5281 [DIS]Updating services using commissioning mode 1
I: 5287 [DIS]CHIP minimal mDNS started advertising.
I: 5291 [DIS]Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1 cp=0
D: 5301 [DIS]Responding with _matterc._udp.local
D: 5305 [DIS]Responding with 64F386FC49D49A8B._matterc._udp.local
D: 5311 [DIS]Responding with 4C3AD5EF3008.local
D: 5315 [DIS]Responding with 4C3AD5EF3008.local
D: 5319 [DIS]Responding with _V65521._sub._matterc._udp.local
D: 5325 [DIS]Responding with _S15._sub._matterc._udp.local
D: 5330 [DIS]Responding with _L3840._sub._matterc._udp.local
D: 5335 [DIS]Responding with _CM._sub._matterc._udp.local
D: 5340 [DIS]Responding with 64F386FC49D49A8B._matterc._udp.local
I: 5346 [DIS]CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 64F386FC49D49A8B.
I: 5356 [DIS]mDNS service published: _matterc._udp
I: 5360 [IN]CASE Server enabling CASE session setups
D: 5365 [IN]SecureSession[0x20001a60]: Allocated Type:2 LSID:58727
D: 5371 [SC]Allocated SecureSession (0x20001a60) - waiting for Sigma1 msg
I: 5377 [SVR]Joining Multicast groups
D: 5381 [ZCL]Emitting StartUp event
D: 5384 [EVL]LogEvent event number: 0x00000000001C0001 priority: 2, endpoint id:  0x0 cluster id: 0x0000_0028 event id: 0x0 Sys timestamp: 0x000000000000013C
I: 5398 [SVR]Server Listening...
I: 5401 [DL]Device Configuration:
I: 5404 [DL]  Serial Number: TEST_SN
I: 5407 [DL]  Vendor Id: 65521 (0xFFF1)
I: 5411 [DL]  Product Id: 32768 (0x8000)
I: 5414 [DL]  Product Name: TEST_PRODUCT
I: 5418 [DL]  Hardware Version: 0
I: 5421 [DL]  Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
I: 5426 [DL]  Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
I: 5433 [DL]  Manufacturing Date: (not set)
I: 5437 [DL]  Device Type: 65535 (0xFFFF)
I: 5441 [SVR]SetupQRCode: [MT:Y.K90AFN00KA0648G00]
I: 5445 [SVR]Copy/paste the below URL in a browser to see the QR Code:
I: 5451 [SVR]https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3AY.K90AFN00KA0648G00
I: 5460 [SVR]Manual pairing code: [34970112332]
D: 5465 [IN]SecureSession[0x20001990]: Released - Type:1 LSID:58726
D: 5471 [IN]SecureSession[0x20001990]: Allocated Type:1 LSID:58728
D: 5477 [SC]Assigned local session key ID 58728
D: 5481 [SC]Waiting for PBKDF param request
I: 5485 [DIS]Updating services using commissioning mode 1
I: 5490 [DIS]CHIP minimal mDNS started advertising.
I: 5495 [DIS]Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1 cp=0
D: 5504 [DIS]Responding with _matterc._udp.local
D: 5508 [DIS]Responding with 712015D598153D77._matterc._udp.local
D: 5514 [DIS]Responding with 4C3AD5EF3008.local
D: 5518 [DIS]Responding with 4C3AD5EF3008.local
D: 5523 [DIS]Responding with _V65521._sub._matterc._udp.local
D: 5528 [DIS]Responding with _S15._sub._matterc._udp.local
D: 5533 [DIS]Responding with _L3840._sub._matterc._udp.local
D: 5539 [DIS]Responding with _CM._sub._matterc._udp.local
D: 5544 [DIS]Responding with 712015D598153D77._matterc._udp.local
I: 5550 [DIS]CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 712015D598153D77.
I: 5559 [DIS]mDNS service published: _matterc._udp
```

So the program execution seems to be good and publish a mDNS service. I also put a static IPv6 address for the device, for now.
I try again to discover it with `Resolve-Dnsname matterc `, and here is the result

[image]

Also, CHIP-TOOL discovering doesn't work. I'm quite stuck because I don't know if my application, or the Zephyr standalone platform, is missing something. @bzbarsky-apple I've seen you are comfortable with commissioning. Any clue on that ? I can provide the application, the .zap, or more logs if necessary. 

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

latest

### Platform

other

### Platform Version(s)

N/A

bzbarsky-apple, 2024-07-31:
> I try again to discover it with Resolve-Dnsname matterc 

That looks like it will try to resolve `matterc.local`, but the relevant bit is `_matterc._udp.local`, no?

> Also, CHIP-TOOL discovering doesn't work.

Please attach, not paste, a log showing:

1. The exact chip-tool command you ran.
2. The log from chip-tool.

QuentinCaldeira-eaton, 2024-08-01:
Hi @bzbarsky-apple, thanks for your response !
I prefer to use dns-sd, so I stopped using Resolve-Dnsname

I attached logs for the application, CHIP-TOOL, and dns-sd. I've succeeded once to go through CASE (but it failed) with dns-sd, but can't reproduce the behaviour anymore (I'm aware that DNS instance name change at each boot). Furthermore, I joined my main app too.  With CHIP_ENABLE_PAIRING_AUTOSTART, I don't need OpenBasicCommissioningWindow because this is done at boot time.

[chiptool.txt](https://github.com/user-attachments/files/16451967/chiptool.txt)
[logdnssd.txt](https://github.com/user-attachments/files/16451969/logdnssd.txt)
[log_app.txt](https://github.com/user-attachments/files/16451975/log_app.txt)
[AppTask.txt](https://github.com/user-attachments/files/16451981/AppTask.txt)

bzbarsky-apple, 2024-08-03:
@QuentinCaldeira-eaton I don't know where the things you are trying to browse for in the log there came from; they have nothing to do with the app_log.txt.

What you want to do with dns-sd is:

1. `dns-sd -B _matterc._udp`
2. If that comes back with an instance name, `dns-sd -L instancename _matterc._udp`

to get the host/port.  And then you could `dns-sd -G v6 hostname.local` to get the IPs.

QuentinCaldeira-eaton, 2024-08-05:
@bzbarsky-apple The things in the logs come from the Matter application logs, but there aren't the same as log_app.txt because the card was reboot between the logs. 

`dns-sd -B _matterc._udp` return nothing, it just timeout. Can this be linked with the fact that my device seems to init a socket with kAny (0.0.0.0) in parameters ?

QuentinCaldeira-eaton, 2024-08-06:
@bzbarsky-apple The only issue I could think of is 
```
E: 103 [IN]IPV6_PKTINFO failed: 109
```

But I don't know if this is critical for commissioning. Is there any CHIP global define (like CHIP_DEVICE_CONFIG_ENABLE_COMMISSIONER_DISCOVERY) is mandatory in order to browse the device over dns-sd. I fix a static ipv6 unicast and multicast address before server init, so mdns should be aware of the ip address before publishing, via Zephyr net_if in Inet. I'm quite stuck because I can't determine if it's an application issue or a platform issue.

bzbarsky-apple, 2024-08-07:
> Can this be linked with the fact that my device seems to init a socket with kAny (0.0.0.0) in parameters ?

No idea, but clearly things are not being advertised properly.

I wish I could help you with the rest of your questions, but I really have no idea.  You might need to add more logging to the relevant codepaths and see where advertising might be failing....

QuentinCaldeira-eaton, 2024-08-07:
@bzbarsky-apple Thanks for your answer. In order to debug that, I've a few more questions : 

- Is there a way to enable more debug information (useful for mdns)
- Is there a plan to support Zephyr as a standalone platform (only based on the bare Zephyr SDK)
- Is mDNS mandatory for commissioning (I think it's a yes)
- Should I indicate more things to the app (variable, functions) to let it know that I'm in a OnNetwork configuration, or just PrintOnboardingCommissionning(kOnNetwork) is sufficient ?
- Is there specific Matter event that are triggered by the OnNetwork connection (like kNetworkChange, etc...) in order to debug each step of the commissioning
- CHIP-TOOL onnetwork command and qrcode command don't seems to do the same thing. Idk if it's normal.
- Why LOGS says that a mDNS service is published (advertise, is that ?) if it's not the case

I will continue to explore the problem and keep you informed

bzbarsky-apple, 2024-08-07:
I don't actually have answers to your questions, as regards your specific platform.

QuentinCaldeira-eaton, 2024-08-08:
> I don't actually have answers to your questions, as regards your specific platform.

I think my main question is now : Do a platform needs a driver for its Network Interface (Thread, BLE, WiFi, Ethernet) in order to override functions in src/include/platform/NetworkCommissioning.h ? It seems that many platform using specific interface provide a WiFi driver, and others like Tizen or ESP provides also an Ethernet Driver. Thread platform is generic as there is an OpenThread platform. 

I don't think the question is platform specific, but it's general. I can't determine if it's mandatory to implement at least the functions of BaseDriver in order to perform a good commissioning. At least, what are the requirement for performing an on network commissioning ?

QuentinCaldeira-eaton, 2024-08-08:
@Damian-Nordic @bzbarsky-apple What do you think of this ? As discussed previously, Ethernet doesn't need specific operations like Scan, etc. But it seems that we need to override the BaseDriver functions (Count, Release etc.) because they are used in the network-commissioning cluster. 

Here is a minimal example : 

```C++
#pragma once

#include <string>
#include <vector>

#include <platform/NetworkCommissioning.h>

#include "CHIPDevicePlatformConfig.h"

namespace chip {
namespace DeviceLayer {
namespace NetworkCommissioning {

class ZephyrEthernetDriver final : public EthernetDriver
{
public:
    class EthernetNetworkIterator final : public NetworkIterator
    {
    public:
        EthernetNetworkIterator(ZephyrEthernetDriver * aDriver) : mDriver(aDriver) {};
        ~EthernetNetworkIterator() override = default;
        
        size_t Count() override { return 1; }
        bool Next(Network & item) override
        {
            if (exhausted)
            {
                return false;
            }
            exhausted = true;
            memcpy(item.networkID, interfaceName, interfaceNameLen);
            item.networkIDLen = interfaceNameLen;
            item.connected    = true;
            return true;
        }
        void Release() override { delete this; }

        uint8_t interfaceName[kMaxNetworkIDLen];
        uint8_t interfaceNameLen = 0;
        bool exhausted           = false;

    private:
        ZephyrEthernetDriver * mDriver;
    };

    uint8_t GetMaxNetworks() override { return 1; }
    NetworkIterator * GetNetworks() override { return new EthernetNetworkIterator(this); }

    CHIP_ERROR Init(NetworkStatusChangeCallback * networkStatusChangeCallback) override;

    static ZephyrEthernetDriver & GetInstance()
    {
        static ZephyrEthernetDriver instance;
        return instance;
    }
};

} // namespace NetworkCommissioning
} // namespace DeviceLayer
} // namespace chip
```

```C++
#include <app-common/zap-generated/cluster-enums.h>
#include <lib/support/CodeUtils.h>
#include <platform/NetworkCommissioning.h>
#include <platform/Zephyr/EthernetDriver.h>

/* Zephyr includes */
#include <zephyr/kernel.h>
#include <zephyr/logging/log.h>
#include <zephyr/linker/sections.h>
#include <errno.h>
#include <stdio.h>
#include <zephyr/net/net_if.h>
#include <zephyr/net/net_core.h>
#include <zephyr/net/net_context.h>
#include <zephyr/net/net_mgmt.h>

namespace chip {
namespace DeviceLayer {
namespace NetworkCommissioning {

CHIP_ERROR ZephyrEthernetDriver::Init(NetworkStatusChangeCallback * networkStatusChangeCallback)
{
    //TODO don't use static mac, but a default one overrided by a CHIP_CONFIG in prj.conf
    uint8_t mac[6] = {0x33, 0x33, 0x11, 0x11, 0x11, 0x11};

    struct net_if *iface;
    struct in6_addr my_ipv6_addr;
    struct in6_addr my_ipv6_mcast_addr;

    iface = net_if_get_first_by_type(&NET_L2_GET_NAME(ETHERNET)); //We only search for ethernet interface
    if (!iface) {
        ChipLogProgress(DeviceLayer, "No Ethernet interface found");
    } else {
        ChipLogProgress(DeviceLayer, "Found an Ethernet interface");
    }
    net_if_set_link_addr(iface, mac, sizeof(mac), NET_LINK_ETHERNET);

        /* Set the static unicast IPv6 address */
    net_addr_pton(AF_INET6, "2001:db8::1", &my_ipv6_addr);

    net_if_ipv6_addr_add(iface, &my_ipv6_addr, NET_ADDR_MANUAL, 0);

    /* Set the static multicast IPv6 address */
    net_addr_pton(AF_INET6, "ff00::1", &my_ipv6_mcast_addr);

    net_if_ipv6_maddr_add(iface, &my_ipv6_mcast_addr);

    return CHIP_NO_ERROR;
}

} // namespace NetworkCommissioning
} // namespace DeviceLayer
} // namespace chip
```

I know I don't have to put static IPV6 addresses in Init, it's just for testing the app. I don't know if Matter need something else in networking. I guess I add a multicast, and a link local address to the ethernet interface. Matter should be able to use it. I can build and flash, but I'm again unable to commissioning.

Damian-Nordic, 2024-08-08:
@QuentinCaldeira-eaton I doubt implementing `EthernetDriver` has anything to do with your DNS-SD issues, so I would suggest focusing on DNS-SD first. As I mentioned before, I'm sorry but I don't have time to help. I would probably start with enabling `CONFIG_NET_SHELL` and trying to manually configure multicast group and checking if you can ping a multicast address on the device.

QuentinCaldeira-eaton, 2024-08-08:
@Damian-Nordic Idk because `EthernetDriver` is directly called by `network-commissioning` cluster, and I've issues with commissioning. I mentionned you because previsously, you said that I need a multicast and link-local address for my device. I would like to know if is the right way to go ?

```C++
/* Set the static unicast IPv6 address */
net_addr_pton(AF_INET6, "2001:db8::1", &my_ipv6_addr);

net_if_ipv6_addr_add(iface, &my_ipv6_addr, NET_ADDR_MANUAL, 0);

/* Set the static multicast IPv6 address */
net_addr_pton(AF_INET6, "ff02::1", &my_ipv6_mcast_addr);

net_if_ipv6_maddr_add(iface, &my_ipv6_mcast_addr);
```

QuentinCaldeira-eaton, 2024-08-13:
Hi @Damian-Nordic and @bzbarsky-apple , I've dig further in the mDNS, and think I find the point of failure : it's in `src/lib/dnssd/minimal_mdns/Server.cpp` when, on line 346
```cpp
if (chip::Loop::Break == mEndpoints.ForEachActiveObject([&](auto * info) 
```

I never enter this condition, and this function always return me the error 0x6c "UNSUPPORTED_CHIP_FEATURE".

Here is my configuration and the error : 

[image]
[image]

The network configuration seems to be good, and my router too because I can commission other matter wifi devices. I don't understand the condition above, and I don't know if I'm doing something wrong here. Any help ? The issue is quite precise now

QuentinCaldeira-eaton, 2024-08-14:
Seems to be a missing Ethernet Driver + x509 from mbedtls

closed 2024-08-14 by QuentinCaldeira-eaton