project-chip/connectedhomeip #34695: Add more tests to the Thermostat presets test for testing the presets constraints- TC_TSTAT_4_2

nivi-apple, 2024-08-01:
# TODO: Add tests for the total number of Presets exceeds the NumberOfPresets supported. Also Add tests for adding presets with preset scenario not present in PresetTypes

nivi-apple, 2024-08-01:
Follow up - have a test exercising the "empty and null are treated identically" behavior, though. for the preset handles. 

test "negative" at all, like change an expected success assertion  for a different status code and see that the way you catch failures leads to expected logs and failure, rather than a different exception? 
Make sure that a really failing product leads to a clean fail of the test.

ReneJosefsen, 2024-08-19:
@ReneJosefsen - Open test plan issues for this and use this issue for making sure we align the scripts accordingly

PR #35141 (2024-08-28): Add test cases for testing additional Presets write and commit constr…
…aints

- Add a test for adding a preset with a preset scenario not present in PresetTypes

- Add a test for testing addition of presets such that the total number of presets added is greater than the total number of presets supported

Fixes: #34695
closed 2024-08-28 by mergify[bot]