project-chip/connectedhomeip #34702: Make the Semantic tag struct global

hicklin, 2024-08-01:
This issue depends on issue #34700.

According to the spec, the `SemanticTagStruct` type should be a globally defined type. See [here](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/f588e50a1bc756f00570a6501b679c18e8b30105/src/data_model/Data-Model.adoc?plain=1#L3274).

Given that now there is a way to represent global structs, we should 
- [ ] move the SemanticTag struct definition in the `descriptor-cluster.xml` to the `global-structs.xml`, without an attributed cluster.
- [ ] Update the NamespaceID type to be the `Namespace` enum from issue #34700.
- [ ] Fix any broken code.

hicklin, 2024-08-01:
There may be an issue here due to the fact that the Mode Select cluster has a struct with the same name. There are a couple of options to consider here
1. Deprecate Mode Select. Mode Base should be an equivalent replacement.
2. Choose a different name. This will cause even more API changes.
3. Fix ZAP to allow for a global and a cluster specific structure with the same name.

On point 3, @bzbarsky-apple did a quick analyses and found that because ZAP selects the "global" name in preference to the cluster-specific one, we'll need an update to zap. Something along the lines of

```
-                        MTRModeSelectClusterSemanticTagStruct * newElement_2;
+                        MTRDataTypeSemanticTagStruct * newElement_2;
```

bzbarsky-apple, 2024-08-03:
Well, and the spec will need to define how these name collisions are handled..

closed 2025-10-08 by mergify[bot]